slic3r/Slic3r #4992: Infinite Loop when infill is 0% and Acoid Crossing Perimiters

nathanshearer, 2020-07-07:
If you set slic3r to use 0% infill, and also enable "Avoid crossing perimeters", then it never finishes generating the g-code.

If the part is effectively solid because the perimeters fill all voids perfectly, then it will never finish generating the g-code regardless of what the infill is set to (0% up to 100%).

For parts that do not require infill, I would like to also have the print head avoid crossing perimeters when possible. Some of my designs can be effectively printed with only thick walls resulting in empty up to 100% solid parts by simply increasing the number of perimeters, and having the slicer generate no infill. I find the prints are faster and have fewer problems.

**To Reproduce**
1. Load up an STL
2. Apply these settings: Seam Position Random, Infill 0%, Avoid Crossing Perimeters
3. Export g-code
4. Watch it never complete

**Expected behavior**
I would expect the export to complete rather quickly, since it would move the tool along internal perimeters for each layer, and only cross a perimeter at a random location when required.

Slic3r Version: 1.3.1-dev build 54a31ee

nathanshearer, 2020-07-07:
[model-and-config.zip](https://github.com/slic3r/Slic3r/files/4887041/model-and-config.zip)

BeatSlayer, 2020-07-13:
Had the same problem before (ACP + mostly perimeters model) and if I rotate the model or change the infill angle it'll slice without problems.