slic3r/Slic3r #4993: Slice to svg error occurred

jxnx888, 2020-07-09:
**Describe the bug**
When I use the "Slice to SVG", there is an error that occurred. Please check the image below.
[image]

I have no idea what is it. I cannot use this feature.

lordofhyphens, 2020-07-09:
Well that's new to me.

Start by passing the model through netfabb:
https://service.netfabb.com/service.php

If that is still throwing errors, you're going to need to post a 3d model (that has been fixed with netfabb) that you see the error with.

jxnx888, 2020-07-10:
There is still the same error, I tried different STL files.
[image]

lordofhyphens, 2020-07-10:
@jxnx888 Do as I asked and attach a copy of a model you are getting the crash on, please.

jxnx888, 2020-07-10:
[hand_ok](https://service.netfabb.com/upload/download.php/hand_ok_fixed.stl?job=86cc499d2f013c07fbead538359b75d32f9e306c)

jxnx888, 2020-07-10:
https://service.netfabb.com/upload/download.php/hand_ok_fixed.stl?job=37d47a91c24e11ea88975254a20161ba