slic3r/Slic3r #4994: Multiple objects to be sliced and printed on dual extrusion- command-line-only

superzaki, 2020-07-16:
Hi all

I want to slice multiple objects 0.obj 1.obj 2.obj... using the command line only; 
And I want to attribute for each object an extruder for printing.
I have been able to do this via the GUI just fine however, thes models are much heavier and the command line is faster based on what I've experienced.
The syntax that I've tried so far goes as:
`slic3r -j 8 --no-gui --load config_all.ini --support-material-extruder 2 --support-material-interface-extruder 2 --dont-arrange --perimeter-extruder 2 --infill-extruder 2 --solid-infill-extruder 2 --perimeter-extruder 1 --infill-extruder 1 --solid-infill-extruder 1 --merge 0.obj 1.obj`
But this results in the combined structed printed using extruder 1 only. Ps. specifying the extruder for the support works just fine.

Many thanks in advance

JPA1238, 2021-04-10:
Hi superzaki,

Do you have any updates on the ability to specify the extruder for certain objects? aka multi-color slicing using CLI.

-JPA-

superzaki, 2021-04-14:
My apologies for not replying sooner..
Unfortunately, I couldn't find any CLI-based tools that allow slicing my objects intuitively for dual extrusion prints. The facts that we were running out of time in the project and the complexity of the 3D objects in question (>80M triangles), have forced me to try Simplify3D. Honestly the tool is a beast and its capability to scale slicing workloads across all 128 cores and >100Gb RAM, is simply amazing.