slic3r/Slic3r #4999: 'Retract Length', 'Lift Z' & 'Retract Speed' Values Are Being Ignored (v1.3.0 & 1.3.1)

rf1k-mjh11, 2020-07-27:
**Issue:**
Some settings that are changed in the GUI dialog box for
'_Printer_', in the section '_Extruder_', in the subsection '_Retraction_'
are ignored during GCode generation and are not reflected in the configuration file, either.

Currently the errors applying to _Retract Length_, _Lift Z_ have been verified, and the bug is presumed to apply to _Retract Speed_, also (but not verified via GCode analysis).
Applicable to **versions 1.3.0 and 1.3.1 dev** (from 20200723).
            Version 1.2.9 **DOES NOT** show the issue (see attached configuration file). 

**To Reproduce**

1. In the _Printer_ settings dialog, under '_Extruder_', '_Retraction_', make the following settings (see screenshot):

- Length:					1.9
- Lift Z:					0.5
- Only lift Z:		Above Z: 0	Below Z: 0
- Speed:					150
- Extra length on restart:		0.0002
- Minimum travel after retraction:	3
- Retract on layer change:		YES
- Wipe while retracting:			YES

2. Generate GCode and export Config file (_File_, _Export config..._)
3. Check applicable settings in configuration file, examine GCode and verify the _Retract_ & _Lift Z_ values that were used.
3.a. Check if the correct Retract Speed was used (this was not verified by myself)

**Attachments**
The STL model used to reproduce the error was a simple 20x20x3mm solid slab (see zip-file attachment).
Configuration files for versions 1.2.9 (**without** bug) and 1.3.0 (**with** bug) have been attached.
[config_v1,2,9_retract_bug_OK.txt](https://github.com/slic3r/Slic3r/files/4983932/config_v1.2.9_retract_bug_OK.txt)
[config_v1,3,0_retract_bug_!_.txt](https://github.com/slic3r/Slic3r/files/4983934/config_v1.3.0_retract_bug_._.txt)
[Slic3r_Retract_Bug_20200723.zip](https://github.com/slic3r/Slic3r/files/4983939/Slic3r_Retract_Bug_20200723.zip)

**Expected behavior**
A changes to any setting via the GUI should be reflected in the generated GCode.

**Screenshots**
Screenshot of setting dialog of v.1.3.0:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 8.1, English, 64bit
 - CPU: AMD Ryzen 3 2200 G, 3.50 GHz, 16 Gbyte memory
 - Slicer versions 1.3.0 and 1.3.1 dev (automated development build from 2020-07-23, time stamped 06:34)

**Additional context**
I am assuming that the Retraction Speed is also not propagating into the GCode generation, but I have not gone to the trouble of verifying the speed used in the GCode that is generated.  The Retraction Speed is shown 'incorrectly' in the configuration file (and at the very end of the GCode file), which is why I am assuming it also has the issue.

lordofhyphens, 2020-07-27:
So, I can't reproduce experimentally as 00e7b009537e865604428bc18159ccb64f11751f (a branch where I was only messing with the brim configuration, so this area of code should be untouched) without wipe. Wipe makes things a bit more interesting due to how GCode handles speed.

lordofhyphens, 2020-07-27:
When you have wipe enabled, Slic3r uses 80% of the travel speed to calculate the wipe speed. This is deliberate. It then plots out however much wipe is necessary based on the travel speed. The final little bit is done at the specified retract speed. 

It then lifts, moves to the next point, and drops down and unretracts at the specified retract/unretract speed.

lordofhyphens, 2020-07-27:
In 1.2.9, the wipe code is on the perl side; however, the calculation is the same as in 1.3.0 and 1.3.1-dev.

https://github.com/slic3r/Slic3r/blob/1.2.9/lib/Slic3r/GCode.pm#L572

We have a limit in GCode in that we don't have the ability to specify two speeds at the same time, and if your extruder axis is hard-limited to a lower speed than 80% of the travel speed the  firmware will slow all axes down to the maximum firmware speed for one of the axes (this is one of the reasons why Slic3r tends not to produce moves with XY and Z moves at the same time).

closed 2020-07-27 by lordofhyphens

rf1k-mjh11, 2020-07-28:
Lord of Hyphens/Joseph,

Thank you for responding so quickly to my bug report (although it is now
closed).

There is still something seriously wrong with it (at least on my
machine):  when I remove the check mark for /Wipe while retracting/, the
configuration file still shows wipe = 1 and the generated GCode also
reflects a wipe operation occurring while retracting.

Additionally, although I couldn't quite follow your three eMail answers
describing the issue and justifying its closure, I cannot see why an
entry into a dialog box should be ignored without any feedback to the
user.  Therefore, at your convenience, please clarify if you can
replicate the above mentioned issue in v1.3.0 (i.e. remove check mark &
check config file).

Thank you,

mjh11