slic3r/Slic3r #5000: Not running on Linux with RX580

tj-murphy, 2020-07-29:
I downloaded Slic3r from https://github.com/slic3r/Slic3r/releases/download/1.3.0/slic3r-1.3.0-linux-x64.tar.bz2 and extracted, then tried to run the 162-byte extensionless Slic3r file. Nothing happened when I launched via the file browser, and when I launched via terminal, I get the following output:
```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
mesa: for the   --simplifycfg-sink-common option: may only occur zero or one times!
mesa: for the   --global-isel-abort option: may only occur zero or one times!
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Segmentation fault (core dumped)
```

Googling these error messages led me to this Issue: https://github.com/slic3r/Slic3r/issues/4645

I was surprised to find another user with my same GPU being unable to run the software. What should I do next to troubleshoot? Happy to provide outputs of any commands or whatever may be helpful.

lordofhyphens, 2020-07-29:
Grab a build from https://dl.slic3r.org/dev/linux/

tj-murphy, 2020-07-29:
I downloaded slic3r-master-latest.tar.bz2 and saw identical results.

lordofhyphens, 2020-07-30:
And the AppImage?

tj-murphy, 2020-07-30:
The AppImage gives results that seem very similar though slightly different.

```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
mesa: for the   --simplifycfg-sink-common option: may only occur zero or one times!
mesa: for the   --global-isel-abort option: may only occur zero or one times!
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/tmp/.mount_Slic3rKMulXj/AppRun: line 65: 957312 Segmentation fault      (core dumped) LD_LIBRARY_PATH="$DIR/usr/lib:${LD_LIBRARY_PATH}" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"
```

lordofhyphens, 2020-08-23:
@tj-murphy  https://github.com/slic3r/Slic3r/issues/4645#issuecomment-451109669

TheCodeExorcist, 2020-09-24:
Hi,
I use a working Slic3r with an RX480, which is almost the same GPU. The AppImage does not work under ArchLinux and gives an output that is somehow close to yours (see #5013) so it could eventually be related. Can you tell us more about your distro and kernel?