slic3r/Slic3r #5001: Can't open/install Slicer on Mac 10.15.4

tahoematt, 2020-08-04:
1.3 Version

MAC OS - 10.15.4 - Operating system type + version

### Behavior
* Can't install/open the app. I get "slicer can't be opened because the developer cannot be verified".
[image]

amelie3103, 2021-02-26:
I tried it on mac and I get the same problem with this message
[image]

lordofhyphens, 2021-02-26:
Going to reiterate how much I hate Apple and everything to do with it.

lordofhyphens, 2021-02-26:
Time to waste a weekend arguing with their signing infrastructure.

bubnikv, 2021-02-26:
And then you will likely find out that OpenGL does not work on the new M1
processor as they deprecated it and there is no way really to debug it.

lordofhyphens, 2021-02-26:
Without the hardware and time to maintain either a fork or port everything to a 3d engine that actually works for Slic3r's needs (really, it needs something both cross-platform supports procedurally-generated assets), MacOS support on the new processor is probably going to be dropped unless @alranel wants to take a crack at it. I don't use Macs at all outside of the bare minimum needed to deal with Slic3r signing garbage.

LangBalthazar, 2021-02-27:
For me this solved it:

**Allowing Third Party Applications to Install on a MacBook**
Open "System Preferences" by clicking on the Apple icon and choosing that option.
Select "Security & Privacy" from the "System Preferences" window.
Select the "General" tab, and select the lock in the lower left corner to allow changes.
Enter your computer username and password, then select "Unlock."

 
[image]

hartos, 2022-02-15:
Control-click the app icon, then choose Open from the shortcut menu. Click Open. The app is saved as an exception to your security settings, and you can open it in the future by double-clicking it just as you can any registered app.