slic3r/Slic3r #5002: [Feature] Injection 3D Printed perimeters

from-nibly, 2020-08-05:
# Purpose

Possible speed and strength improvement to 3D prints.

# Basic Concept:

This would turn perimeters into walls that resemble a form mold concrete wall. Two walls would be printed separated by a gap of some sort, which would then be filled every N number of layers with a large amount of extrusion.

# Details:

I read a brief [article](https://www.fabbaloo.com/blog/2020/8/3/injection-printing) about a new type of 3D printing strategy where the researchers printed with 100% infill with much shorter time by using this method.  The idea would be to utilize this strategy with perimeters only, achieving something with less than 100% infill.
[image]

The diagram from the article (above) shows the basic concept. You print "mold" walls that alternate in thickness creating little teeth that the solid infill can grab onto. Then every N number of layers you fill in the "mold" with plastic. Doing this while filling in the "mold" every 4 layers would potentially save 3/12 or 1/4 the total travel time (assuming of course you can fill in the gap at the same speed you print the walls). Plus with the teeth and the solid infill the layer adhesion should be a lot stronger since the layer adhesion would no longer be only planar across a single axis. This should also require no additional plastic over a design using 3 perimeters.

# Considerations:

- In the article they describe temperature and geometry issues. 
- This obviously wouldn't work at all in places where the overhang of a wall exceeds that of the requirement for printing supports, as one of the walls would simply collapse onto the other. 
- This also wouldn't work for base or top perimeters since you can't print both sides of the "mold" without blocking the print head from being able to fill it.
- This also requires that your hotend can heat the plastic fast enough to essentially print multiple layers at the same time. Depending on how many layers you are printing that could be a lot of plastic.
- Depending on how high the "mold" walls are this could produce a lot of general instability while printing resulting in a lot of failures.

I am a software developer but I basically have absolutely no experience with programming geometries, or slicers, or anything real with c++ so unfortunately this is just a heads-up kind of feature request, but I am willing to help in any way I can with testing, or discussion.

lordofhyphens, 2020-08-05:
Seems like this is very similar to how gap fill is modeled in Slic3r.

from-nibly, 2020-08-05:
@lordofhyphens Exactly, The extrusion could have either a heck of a lot of melt to produce a fluid pour or less of a melt for kind of a spaghetti/silly string into the cavity, There are likely advantages and disadvantages to various filling strategies. A setting could be set for mold fill percentage which would change the rate of extrusion vs the rate of travel over the gap producing different kinds of infill; spaghetti vs totally fluid pour.

from-nibly, 2020-08-11:
The other thing I'm realizing is that if you did 5 perimeters and filled the second and fourth perimeters with injection you can offset the height of the second perimeter injection by a couple layers. This would remove all continuous layer shear points, as there would be no single continuous z plane that would not intersect solid plastic.  This strategy could also be used with any odd number of perimeters.