slic3r/Slic3r #5005: can't be opened because the identity of the developer cannot be confirmed

gitcnd, 2020-08-13:
**Describe the bug**
Malware-protection and user safety are EXTREMELY IMPORTANT to everyone, especially Apple, which is why they now ban lazy couldn't-care-less developers from shipping untrusted code that runs on our machines.  Anything unsigned can easily be hijacked and trojanned, and in case you're not aware, properly signed code is inspected by apple for risks that could compromise our systems.  It protects us against your own mistakes, and many of your possible malicious intents, and against hackers leveraging your laziness to attack us.

**Expected behavior**
Developers spend the effort to keep us safe by properly signing their code.

**Note**
Please do not close this without fixing.  I know this sounds rude, but think about how rude you're being by forcing all your users to disable their security and the unnecessary risks you're placing on all your customers.  Which is worse?

[image]

lordofhyphens, 2020-08-13:
I have pinged Alessandro, as I am dealing with a local natural disaster
and barely have any access to anything. I believe that the copy we have up
on our CI server (everything is signed and has been signed) has expired.

lordofhyphens, 2020-08-23:
Yep, have to notarize everything because Apple. I really hate this platform.

ArchakGoel, 2024-10-02:
Having the same issue with Sonoma 14.4.1 and slic3r-1.3.0.dmg