slic3r/Slic3r #5007: Slic3r command line --cut-x command not working

simar1998, 2020-08-13:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

Documentation specifies that a user can cut a model along the x and y axis using the --cut-x X command from the command line. However this does not work. If i were to run the command `Slic3r  xyzCalibration_cube.stl --cut  10', Slic3r sucessfully cuts the folowing mesh file 10 mm on the z axis. However when the command 'Slic3r  xyzCalibration_cube.stl --cut-x  10' is run from the command line, Slic3r returns with the following message 'Unknown option: cut-x'. Am I invoking it wrong ?. According to the following document [link](https://manual.slic3r.org/advanced/command-line)  it states that --cut-x X, --cut-y Y: cuts the given model(s) in two halves along the X or Y axis at the given absolute coordinate and exports both resulting models.

I have attached the thingiverse linkg to the mesh file I am using to do this [link](https://www.thingiverse.com/thing:1278865) I am using Slic3r 1.3.0, on Ubuntu 20.04.1 LTS.

lordofhyphens, 2020-08-13:
Use 1.3.1-dev

simar1998, 2020-08-14:
@lordofhyphens I attempted using SLic3r 1.3.1-dev  however the results are exactly the same, There is not difference, I still get the same error message.

simar1998, 2020-08-14:
I downloaded the dev version of slic3r from [here](https://dl.slic3r.org/dev/linux/) and when it is run and the --cut-x command is submitted it shows the error listed above. However I cloned the Slic3r project, compiled the debug version myself and within it it seems to work. However I must point out that all executables and prebuilt versions of slic3r that I have tested do not have this feature working.