slic3r/Slic3r #5008: Can you select a preset from the cmd line?

gilesp1729, 2020-08-18:
I notice that programs like RepetierHost can:
- fill a list of Slic3r presets, both system and user, with the available presets for print, printer and filament
- select a Slic3r preset from a list
- fill a custom .ini file and slice with it

all without having the save the config in the GUI beforehand. Now there doesn't seem to be any cmd line options that allow you to do these things. Is this possible without me having to write code to parse slic3r.ini and <vendor>.ini myself? (I have not investigated libslic3r yet, so let me know if it's in there and I'll go away and have a look)

lordofhyphens, 2020-08-18:
It's making its own files and submitting them to a CLI Slic3r for the most
part.

gilesp1729, 2020-08-18:
Repetier is querying the Slic3r presets, listing them in its own window, allowing you to choose one, and optionally overriding some settings. When I look at the .ini file Repetier builds up, it's an exact copy of the settings in the vendor ini file, so it has either parsed the .ini file(s) itself or called on something in Slic3r to do it. It would make sense for Slic3r to do this, as the format of the ini files is kind of internal and may change (though it seems simple enough).

gilesp1729, 2020-09-03:
User presets (in both vanilla Slic3r and PrusaSlicer) are held in separate ini files, but it seems system presets are a Prusa-specific thing. They have a more complicated ini file structure with inheritance. Parsing their ini file seems to be the only way forward for now, so I'll close this issue.

closed 2020-09-03 by gilesp1729