slic3r/Slic3r #5011: Translation - PT-BR

GIBA-BDV, 2020-08-26:
Hi, 
My name is Giovanni, I'm a Brazilian mechanical engineer.

I'm not used to program or to develop any software, but I want to contribute with the community.

In my graduation, I studied the 3D printing universe and love it. But in my country, we don't have slicers translated in Brazilian Portuguese as well as national low cost 3D printers.

I want to develop a low cost 3D printer, based on the RepRap documentation and use a open source slicer to create the printing profiles, like Prusa does.

So, I want to contribute to the project with a translation to my native language.

Can someone help me? Can someone teach me?

I don't know how Github works, so I'm typing a issue, but I don't think it's right. Sorry for my English, I'm not fluent.

foreachthing, 2020-08-26:
Sorry to hijack this issue: use [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) ([Website](https://www.prusa3d.com/prusaslicer/)). It already has PT-BR included and it's based on Slic3r.:
[image]

TheCodeExorcist, 2020-09-24:
I think a good start would be to look at [the ".po" files located in the "Translation" folder](https://github.com/slic3r/Slic3r/tree/master/translation). The files along with the commits and pull requests done on them should give you a good idea on how to do it, have fun ;)

GIBA-BDV, 2020-10-21:
Cool, I will look at the translation folder. Thanks to the support.