slic3r/Slic3r #5012: when i try to run Slic3r on ubuntu 20.04 i get core dump

MartinEtcheverry, 2020-08-26:
when i run perl ./slic3r.pl $* i get this:
assertion 'size >= 0' failed in GtkScrollbar
Violación de segmento (`core' generado)

lordofhyphens, 2020-08-26:
Version? Is this one of our builds?

MartinEtcheverry, 2020-08-26:
yes i downloaded today.

lordofhyphens, 2020-08-26:
@MartinEtcheverry Please look this over and fill out the necessary information.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

MartinEtcheverry, 2020-08-26:
Describe the bug:
when i run perl ./slic3r.pl $* i get this:
assertion 'size >= 0' failed in GtkScrollbar
Violación de segmento (`core' generado)

to reproduce:
git clone https://github.com/alexrj/Slic3r.git
cd Slic3r
git checkout -b origin/master
export LDLOADLIBS=-lstdc++
perl Build.PL
perl Build.PL --gui
perl ./slic3r.pl $*

Expected behavior
just run 

Desktop
. OS: Linux
. Ubuntu 20.04

Version of build : 1.3.0-1529-g92abbc42

Patola, 2020-08-26:
If you allow me to say something in this conversation, @MartinEtcheverry, have you followed the entire compilation instructions for Ubuntu and ensured you have all libraries, program, scripts necessary to build slic3r? It's quite a long list. https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

MartinEtcheverry, 2020-08-26:
yes , some libraries are updated versions (beacuse is ubuntu 20.04) and the text is for ubuntu 16.04  ...but yes . and it compiles fine. Yes i believe that is a problem with some library , but i don't have the knowledge in how to pinpoint it. 
Belive me , i don't want to bother to anyone , but i am in a "no movement left" situation, well except reinstall the system and try to install again.
and by the way , thanks to all to take time to respond my messages. i really apreciate that.