slic3r/Slic3r #5013: Slic3r Dev AppImage Fails to Open GUI (Arch)

mcdviii, 2020-08-28:
Slic3r Dev AppImage Fails to Open GUI (Arch)

### Version
1.3.1-dev (AppImage) f89b950e8d8754a65e4bb26bc24ccad2

### Operating system type + version
Arch Linux

### Behavior
```
$ ~/Source/AppImage-Binaries/Slic3r-master-latest.AppImage          
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'GLXBadContext'.
  (Details: serial 842 error_code 158 request_code 152 minor_code 6)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.
```

TheCodeExorcist, 2020-09-24:
Confirmed, same problem with all Slic3r AppImages i could test on Arch. The only difference in terms of output is because i have a radeon graphics card so mesa loader returns "failed to open radeon" instead of "failed to open iris". I thought at first that it was an issue related to my graphics driver but it appears to be more Slic3r related as it also appears with other graphics drivers from other vendors.

TheCodeExorcist, 2020-09-24:
Also, it does not seem to be kernel-related as my workstation uses 5.6.19-rt12-1-rt.

lordofhyphens, 2020-09-24:
https://github.com/slic3r/Slic3r/issues/4645#issuecomment-451109669

TheCodeExorcist, 2020-09-24:
I tried the procedure described in the comment, but in my case this file doesn't exist:
squashfs-root/usr/lib/swrast_dri.so 

Tested with today's Slic3r-master-latest.AppImage

5rxp, 2022-03-23:
Same issue on Ubuntu 20.04

alexban011, 2022-05-31:
Still happens on archlinux 5.18.1-arch1-1