slic3r/Slic3r #5014: Spot welding layers for additional strength

boy1dr, 2020-08-29:
While a 3D printer is putting layers down it is placing hot plastic on cold plastic a lot of the time.  
The idea here is to occasionally press the nozzle in to the job fusing several layers together with the hot nozzle, creating an internal vertical structure for added part strength.  
  
I have tested this and it works  
https://github.com/boy1dr/fdm_spot_welded_layers  
https://www.youtube.com/watch?v=Om2sPvlqiZs  
Although i never did get around to being more scientific about testing the result, it was harder to snap the spot welded part.
  
### Implementation
It should be an expert feature.  
Let the user supply density, depth and dwell values.  
Given the surface area of any layer the slicer could determine how many points to spot weld.  
These points should be available in a grid so they line up vertically within the perimeters (grid spacing is the density).  
Spot welding should occur close to the edge of the inner perimeter.

This was the welding strategy for my test piece.  
[image]

  
### Concerns  
Nozzle will have to protrude from the print head by at least the depth value.  
Nozzle must be conical.  
Print material cannot be too sticky.  
  
Thanks for reading, feel free to close this off if it's more trouble than it's worth ;)

lordofhyphens, 2020-08-29:
So is this moving a hot nozzle and intentionally overextruding a little after the perims are laid down?

boy1dr, 2020-08-29:
@lordofhyphens it's just pressing the nozzle in to the print for a moment at strategic spots. there's no extrusion occurring while it does this.

boy1dr, 2020-08-29:
in my tests i extruded a layer, did the spot welds then moved on to the next layer.
Here's the GCode for the 4 spot welds. taken from the test.gcode file on my github. this was repeated after each layer finished.

G10 ;retract
G0 X50.8 Y64.2 F3600
G0 Z0.9
G4 P200
G0 Z1.5
G0 X64.2 Y64.2 F3600
G0 Z0.9
G4 P200
G0 Z1.5
G0 X64.2 Y50.8 F3600
G0 Z0.9
G4 P200
G0 Z1.5
G0 X50.8 Y50.8 F3600
G0 Z0.9
G4 P200
G0 Z1.5
G0 Z1.5
G0 X50 Y50 F1800
G11 ;unretract