slic3r/Slic3r #5017: 3D view quarter size

oemupenn, 2020-09-03:
**Describe the bug**

The 3D viewport is a quarter of the size of what it should be, but to interact with objects you have to use the mouse as if the viewport is regular sized.

**Screenshots**

https://encrptd.com/image/eZx9

**Desktop (please complete the following information):**
 - Ubuntu 20.04
 - Version 1.3.0

**Additional context**

Slic3r from Ubuntu repos
Gnome Shell from Ubuntu repos
Nvidia 450 driver from graphics-drivers PPA

lordofhyphens, 2020-09-04:
Well that's a new one. 

Can't reproduce on my Debian sid system presently.

Maxwellfire, 2020-10-21:
I have also seen this on a windows system on a high res display, but not reproducibly, and it hasn't happened in a while

janput, 2021-05-06:
I have the same issue on Ubuntu 20.04.2 LTS with version 1.3.0
The issue only appears when i set the desktop scaling to anything different from 100%
I use a 4K display and my default scale is 125%