slic3r/Slic3r #5018: Unable to configure Slic3r in RepetierHost under Linux [Linux][RepetierHost]

mjscosta, 2020-09-04:
**Describe the bug**
Followed installation procedure for RepetierHost under Linux, but when hitting the configuration button the following error was printed in the log:

`13:22:35.691 : <Slic3r> /home/******/.local/share/RepetierHost/Slic3r/Slic3r: line 7: /home/******/.local/share/RepetierHost/perl-local: No such file or directory`

**TLDR**
The issue seems to be related with the `Slic3r` bash wrapper script, that fails to retrieve the script directory to set the context for the execution. Replacing the line `BIN=$(readlink "$0")` by `BIN=$(readlink -f "$0")` in the script seems to fix the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Download slic3r and install it locally
```
cd /tmp/
wget https://dl.slic3r.org/linux/slic3r-1.3.0-linux-x64.tar.bz2
tar xjf slic3r-1.3.0-linux-x64.tar.bz2 
mkdir ~/opt ; mv Slic3r/ ~/opt

```
2. Launch RepetierHost '....'
3. Goto Slicer/Manager and add a slic3r slicer
[image]
4. Go to Slicers, select *slic3f* from the combo-box, and press the *Configuration* button.
[image]
Error is displayed in the log area.

**Expected behavior**
Configuration would display with no errors.

**Screenshots**
See above.

**Desktop (please complete the following information):**
 - OS: Linux, Ubuntu 18.04 LTS
 - Version: 1.3.0

**Solution**
In the bash script `Slic3r` packaged with the distribution replace `BIN=$(readlink "$0")` by `BIN=$(readlink -f "$0")`.

lordofhyphens, 2020-09-04:
We don't support RepetierHost and never have.

mjscosta, 2020-09-15:
Hi @lordofhyphens , I've posted the issue because I believe it to be a bug in the wrapper that can happen from any calling application, but as it seems I have not been able to reproduce.

Anyway thanks for the info!

mjscosta, 2020-09-15:
If you agree then, I'll close the ticket.

closed 2020-09-24 by lordofhyphens