slic3r/Slic3r #5019: slic3r hangs forever with 200% CPU when attempting to slice

jin-eld, 2020-09-15:
I have a model that does not slice properly if particular settings were applied. I did not identify what settings exactly influence it, but I am attaching a working and a non working .ini

This happens on master HEAD (92abbc42dfdd5385c1f9c3a450e2f3da835f8b8d)

**Describe the bug**

Load the model and slice it - the progress dialogue will appear and just hang there, CPU load goes up to maximum. I tried waiting for a while, but slic3r did not recover.

This does not happen on slic3r-1.3.0-13.fc31.x86_64 - there the model with the given settings does slice correctly.

**3D Model and Slic3r Configuration Export**
[bug.zip](https://github.com/slic3r/Slic3r/files/5226934/bug.zip)

**Expected behavior**
Successful slicing to a g-code file.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: master HEAD at the time of writing / 92abbc42dfdd5385c1f9c3a450e2f3da835f8b8d

**Additional context**

working.ini does slice the model on the master HEAD revision
non-working.ini exposes the error

non-working.ini slices fine with the released 1.3.0 version

dwillmore, 2020-09-15:
Wow, I'm looking through the differences and I'm just not seeing anything that sticks out.  Anyone know a quick way to do a binary merge?

jin-eld, 2020-09-15:
I think I know what causes it. When I have "Minimum shell thickness" set to zero, slicing works. When I set both values (vertical and horizontal) to 2mm - it will hang when slicing.

dwillmore, 2020-09-16:
Verified here.  Used Slic3r-1.3.1-dev-92abbc4-x86_64.AppImage on Fedora 32 with my profile (similar to @jin-eld's profile) and it hangs.  Restarted it, changed both shell thicknesses to 0 and resliced.  Worked just fine.  Changed them back and resliced--hang.

It's just a bisect to find out the commit.  I wonder if the cli version has this problem as well.  I've never used it, but it would be a lot easier to use it to bisect than the gui.

joluxer, 2021-03-14:
Found a simelar bug here, where the slicer hangs, when the horizontal shell thickness is not an integral multiple of the layer height and sometimes also also when the vertical shell thickness is not an integral multiple of the extrusion width.

lordofhyphens, 2021-03-14:
I found a bug in the horizontal (and you can trigger it with the CLI using `--load`).

The feature was computing `std::abs(min_shell_thickness - current_shell_thickness) > epsilon` as a loop conditional, which meant that if you had an oddball layer height it could shoot right over -epsilon < 0 < epsilon and then continue forever.

lordofhyphens, 2021-03-15:
@jin-eld @dwillmore There was an issue around the top/bottom minimum shell thickness where the check for shell thickness can skip completely over the threshold (epsilon). 

There's a fix now available in a PR.

commit af434695 (2021-03-15): Throw a runtime error if the number of solid layers grows above the total possible layers.
Edit the loop syntax to need the current shell thickness to be less than the minimum to add more layers (avoid infinite loop).

Add test to cover normal operation and regression for this layer arrangement.

Fixes #5019
PR #5064 (2021-03-15): Fix issue around infinite loop in min vertical shells.
Throw a runtime error if the number of solid layers grows above the total possible layers.

Edit the loop syntax to need the current shell thickness to be less than the minimum to add more layers (avoid infinite loop).

Add test to cover normal operation and regression for this layer arrangement.

Fixes #5019
commit c61c25b1 (2021-03-15): Throw a runtime error if the number of solid layers grows above the total possible layers.
Edit the loop syntax to need the current shell thickness to be less than the minimum to add more layers (avoid infinite loop).

Add test to cover normal operation and regression for this layer arrangement.

Fixes #5019
closed 2021-03-15 by lordofhyphens