slic3r/Slic3r #5021: Feature request: Templates for scaling

sonrixsonrix, 2020-09-24:
I have a delta printer that has tiny deviations in X-Axis and Y-Axis sizes.
I have tried to find the reason and calibrate. But it is not easy to do this on a delta printer.
While you could solve this in a few minutes on a carthesian style printer, there is no easy fix for this on a delta.

I can scale the print object in slic3r and get perfect accuracy.
But doing so requires entering two menues (scale-x and scale-y) and filling in the same values for every print object.

It would be a great help if you could add an option in the printer menu, to automaticaly apply individiual X-Y-Z scaling values when adding a new object to the plater.

A checkbox, if it shall be active and a table for the 3 values would be enough. The functionallty is already there, there would only be 3 function call required. And a short routine checking if the feature is active when you do the Adding of objects.

Thanks for your help. I would program it myself, if slic3r was in any language I know...

supermerill, 2020-09-24:
You can use the filament shrinkage setting for that.

sonrixsonrix, 2020-09-24:
Please explain a bid more, how I can decrease X size by 2.1 percent and y size by 1.7 percent with filament shrinkage settings?

Gesendet: Donnerstag, 24. September 2020 um 14:31 Uhr

supermerill, 2020-09-24:
Sorry, I've deleted my message in the meantime, because I confused the repository with another. I was wrong.