slic3r/Slic3r #5024: Option for [Bridging] Infill to start on supported areas [first] - work outward.

bkgoodman, 2020-10-07:
Take the following print as an example. I'm obviously trying to avoid supports, but have a very small overhang:

[image]

The way it prints (even when selecting "infill before perimeter") is to draw-out the bridge infil as shown (left to right) (then the perimeter). The problem is, when doing so, it is not drawing on anything and it gets a bit wonky.

**Describe the solution you'd like**
Rather than a straight "left to right"-type infill pattern - if it were to start in the supported areas, do these first, and then work outward to the unsupported areas (then the perimeter) this infill would work properly. If done literally as I describe, the infill in this case would basically become a spiral that starts in the center of the post and works outward.

Even better - having this as a "modifier" that would allow me to apply this rule to specific sections of the design would be optimal. 

I think this would be a widely useful enhancement. 

**Describe alternatives you've considered**
Adding supports - and obvious stuff to avoid supports

lordofhyphens, 2020-10-07:
This is more complicated than it looks; every layer path is computed in
parallel. You need to know what is actually underneath it (computed layer
paths) in order to know if a sub-area is truly unsupported.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bkgoodman, 2020-10-08:
Even an option to do this ONLY for/start-on a certain layer would be _extraordinarily_ helpful. This would go a tremendous away to improving print quality while removing supports.