slic3r/Slic3r #5025: [FR] add "Before Retract Gcode" and "After Retract Gcode" sections to "Printer Settings - Custom G-Code"

Quas7, 2020-10-10:
Slic3r 1.3.0

**Is your feature request related to a problem? Please describe.**
Many TMC stepper drivers feature an UART/SPI interface that allows on the fly motor current and operation mode (Stealthchop vs. Spreadcycle) adjustment. Marlin 2.0.x provides the appropriate commands to apply during printing.
For rapid moves, spreadcycle with optionally even increased current would be beneficial.
Especially these high speed moves are required for geared extruders during retracts and un-retracts.
During normal printing, lower current and stealthchop would be prefered to reduce waste heat and noise.

**Describe the solution you'd like**
Add "Before Retract Gcode" and "After Retract Gcode" fields to "Printer Settings - Custom G-Code" to use:
M569 - Set TMC stepping mode
M906 - TMC Motor Current

**Describe alternatives you've considered**
straight forward python post-processing of gcode files looking for the specific "G1 E{retractmm} F{retractspeed}" command and adding above below the TMC commands mentioned above.

Also, it is possible to use "hybrid stepping" in Marlin for the extruder:
 #define E0_HYBRID_THRESHOLD   40 //switch to spreadcycle
But, this does not increase the motor current simultaneously (but could be added - will post FR in Marlin as well).

**Additional context**
https://marlinfw.org/docs/gcode/M906.html
https://marlinfw.org/docs/gcode/M569.html

dwillmore, 2020-10-10:
I would suggest pursuing the Marlin firmware modification as your best
course of action.

lordofhyphens, 2020-10-10:
Agreed; I think this is out of scope of Slic3r itself.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.