slic3r/Slic3r #5028: big slicing error = CRITICAL

gratefulfrog, 2020-10-27:
**Describe the bug**
The slic3r 1.3.0 does not create proper gcode for the attached stl. The hollow part of the column is taken as the outside not the empty inside. 
I have used Slic3R for 5 years and never witnessed such an issue

**To Reproduce**
Steps to reproduce the behavior:
1. Load the file: phoneMount.stl
2. click "preview"
3. observe the column below the ball as solid and too small. It is the empty part not the external skin!

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.
[Slic3r_1.3.0_bug.zip](https://github.com/slic3r/Slic3r/files/5446786/Slic3r_1.3.0_bug.zip)

**Expected behavior**
I expect the sliced version to be exactly as the original STL file.

**Screenshots**
1. 3D Model view.png shows the model after loading
2. Preview_view.png shows the preview of slicing results. NOTE: the diameter of the base column is a visualization of the bug!

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu  18.04
 - Version : 1.3.0

**Additional context**
Slicing is correct if I cut a tiny but off the top of the model and keep the lower portion. Cut height at 25.0 for example.

Thanks for your help - I love the tool!
Bob

dwillmore, 2020-10-27:
Could you include the model file itself?

lordofhyphens, 2020-10-27:
Yeah, the model itself isn't present, and that's necessary.

gratefulfrog, 2020-10-27:
oops sorry
Here it is.
Or shall I make a new zip??

Please help, I cannot slice this model, it also fails on version 1.3.1. On
1.2.9 it slices, but I cannot make a brim....

gratefulfrog, 2020-10-27:
Sorry for my error!
[Slic3r_1.3.0_bug_with_model.zip](https://github.com/slic3r/Slic3r/files/5446918/Slic3r_1.3.0_bug_with_model.zip)

dwillmore, 2020-10-27:
What did you create the model in?  OpenSCAD?  I'm guessing that the triangles in the lower shaft--on the inside--are backwards.

dwillmore, 2020-10-27:
Yep, that looks like what the problem is.  What did you generate the model with?

gratefulfrog, 2020-10-27:
OPENSCAD 2020.10.11.nightly (git b416aa2)

The model displays fine in Meshlab, and slices fine in Cura and Prusaslcier
as well as in Slic3r 1.2.9 (although there are other issues in that version)

Suddenly, changing nothing, it sliced properly, but only once, now it fails
again.

Any thoughts?

lordofhyphens, 2020-10-27:
Doing the cut causes the mesh to be regenerated.

Slic3r is picky about garbage input models, and making it more tolerant of
garbage input has never been a priority. I have no idea what Prusa's team
changed if it "works fine" there.

gratefulfrog, 2020-10-27:
I see. 

What about this model, are the triangles correct, in your opinion?

It was generated with the stable version of openscad from 2019.
[phoneMount_openscad_stable.stl.zip](https://github.com/slic3r/Slic3r/files/5447191/phoneMount_openscad_stable.stl.zip)

dwillmore, 2020-10-27:
Can you share the source file?

lordofhyphens, 2020-10-27:
https://service.netfabb.com/service.php

First thing I would do is pass the model through that service in these
cases.

dwillmore, 2020-10-27:
Looking at the mesh, it looks like it left the outside of the spherical part of the head burried between it and the cylendrical portion.  That's not a proper mesh.  Those triangles are 'inside' the mesh.

Sorry that it occasionally takes me a while to reply--github is telling me 'You can't comment at this time.' from time to time for no reason I can understand.

gratefulfrog, 2020-10-27:
Thanks for your help!
I really appreciate it!
Here's the openscad source.
[phoneMount.scad.zip](https://github.com/slic3r/Slic3r/files/5447326/phoneMount.scad.zip)

gratefulfrog, 2020-10-27:
Indeed, when fixed with netfab, it slices properly! Thanks so much!!!

dwillmore, 2020-10-27:
You are missing a union() in the BallandRod() function.  You aren't properly joining their meshes, just stuffing them both into the mesh.  Add a union(){ at line 38 and a } at 42 and it works fine.

gratefulfrog, 2020-10-27:
Hi @dwillmore,

I'm not sure that fixes it. I tried with the nightly version of openscad
and it fails still. I'm thinking that there is a bug in the nightly. And
will use the stable version from now on! In that case, there is no need for
the union() - I believe it iis implicit in all the outputs, isn't it?

Cheers,
Bob

dwillmore, 2020-10-27:
The release version of OpenSCAD has an implicit union() there, but the nightlies do not.  So, to be correct, it is needed, but the release versions will do it for you.  If you use the nightlies, you have to be careful to code things correctly.

gratefulfrog, 2020-10-27:
Mmm.  Well, with the nightly and the union() I still cannot slice on the
1.3.0 version...

I will temporarily go back to the stable version of openscad, which has
other known problems, to get this part to print!

Thanks again!!!!

dwillmore, 2020-10-27:
Okay, I'm looking at your openscad code and you're doing a lot of things that are known to cause problems in the generated mesh.  We can discuss it here or elsewhere if you prefer.

gratefulfrog, 2020-10-27:
I would love to learn from your expertise, if you have the time? I really
appreciate the help!

If you would prefer to take this to a private discussion, we can message on
github or even email? as you like...

dwillmore, 2020-10-27:
Take a look at this and see if it makes sense to you how/why I changed it.  It renders a good deal faster as well.
[phoneMount_edited.zip](https://github.com/slic3r/Slic3r/files/5447811/phoneMount_edited.zip)

Basically what I did was to get rid of the intersection.  I can see how you wanted to make the ball and rod meet at a surface, but that's computationally problematic.  It's much better to have them overlap each other and use a union() on them.  Ideally, you want to do all of your unions() and then a difference().  I like to order my designs as a union of the outside shape differenced with a union of the inside (removed) shape.  For your design that would be:
`difference(){
  union(){
    ball();
    rod();
  }
  union(){
    nut();
    hollow();
  }
}`

That will give you the fastest code and the most stable mesh generated, IMHO.  I am not an expert.  I've just used OpenSCAD on and off for a few years for various fun projects.

gratefulfrog, 2020-10-27:
I see! 
Thanks for the tips!  
I understand how the intersection is a risky call and slow to process as well.

Really thanks so much!!!

Here's my version of the code, in your style, which I like, and the stl generated which slices perfectly!
[Dwilmore_Style.zip](https://github.com/slic3r/Slic3r/files/5447956/Dwilmore_Style.zip)

I'm sorry for this bug report which has nothing to do with slic3r !  I'm closing it ;-)

Cheers,
Bob

closed 2020-10-27 by gratefulfrog

dwillmore, 2020-10-27:
No problem, glad to be able to close an issue. :)  Only one problem with your code. ;)  willmore has two 'L's in it. :)

gratefulfrog, 2020-10-27:
Oh no!

I'm so sorry for your name!

Here's an update! :-D
[Dwillmore_Style.zip](https://github.com/slic3r/Slic3r/files/5447982/Dwillmore_Style.zip)

dwillmore, 2020-10-27:
One more bit of feedback from the openSCAD developers.  The code they use for union() may be slow, so maybe don't do a union of the inside parts:
```
difference(){
 union(){
    ball();
    rod();
  }
  nut();
  hollow();
}
```

It's not as pretty, but it may help when performance matters--like with parts much more complex than yours.  Keep that in mind when following my structuring in the future.  If the code ever starts getting too slow, take that last union() out.