slic3r/Slic3r #5030: Slic3r crashes with gyroid infill in Linux

Jan-NiklasB, 2020-11-24:
**Describe the bug**
When adding a stl model to Slic3r AppImage (1.3.0) and trying to preview it with infill set to gyroid with density 20% Slic3r crashes with following message:

`terminate called after throwing an instance of 'ClipperLib::clipperException'
  what():  Coordinate outside allowed range
/tmp/.mount_Slic3r43EV16/AppRun: Zeile 11: 15862 Abgebrochen             (Speicherabzug geschrieben) LD_LIBRARY_PATH="$DIR/usr/lib" "${DIR}/usr/bin/perl-local" -I"${DIR}/usr/lib/local-lib/lib/perl5" "${DIR}/usr/bin/slic3r.pl" --gui "$@"
`

I'm using Linux Mint Cinnamon 19 with Kernel 4.15.0 and also saw this behaviour on desktop PC with a older version of Linux Mint Cinnamon (18 if I remember right).

**To Reproduce**
Steps to reproduce the behavior:
1. Add STL model (60 x 30 x 5 mm in my case)
2. Change infill pattern to "gyroid"
3. Change to "Preview" tab
4. Program crashes

**3D Model and Slic3r Configuration Export**
[Model_config.zip](https://github.com/slic3r/Slic3r/files/5590584/Model_config.zip)

**Expected behavior**
Seeing the preview of the model with gyroid infill as it does in windows

**Screenshots**
-

**Desktop (please complete the following information):**
 - OS: Linux Mint 19, Kernel 4.15.0-112generic    Cinnamon 3.8.9
 - Version: 1.3.0

alagopus, 2020-11-25:
Reproducible with Slic3r-1.3.0-x86_64.AppImage both adaptive and fixed layer height. Happens as well with 
Export G-code even if you skip preview.

Ubuntu 20.04.1 LTS
Budgie Desktop
AMD Ryzen 7

Jan-NiklasB, 2020-11-25:
Maybe it's worth to add, that I also have an AMD Ryzen 1700X, for the case that it's a CPU dependent bug.

lordofhyphens, 2021-03-15:
Can't reproduce with https://dl.slic3r.org/dev/linux/Slic3r-1.3.1-dev-18ddd3a-x86_64.AppImage

closed 2021-03-15 by lordofhyphens