slic3r/Slic3r #5031: Add support for selective heatbed heating to reduce energy consumption and save the earth ;). Actual there is no hardware but you can imagine that a kind of checkerboard pattern heatbed is not too complicated. What do you think?

matzejo, 2020-11-30:
### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ProjectPatatoe, 2020-12-09:
just a few quick thoughts:
pros: 
* possibly quicker heating (more power available? probably not in most cases)
* less power consumption
* easier on the mosfets
cons:
* uneven heating
* uneven flexing of bed from expansion
* less predictable changes in bed between different models

matzejo, 2020-12-09:
Good points. Here are some comments: 
- uneven heating  
'yes of course ;)
- uneven flexing 
'yes maybe for this magnetic flex bed but a glas bed is more stable. Glas expansion on 1m  lenght from  0° to 50° ~0.5mm