slic3r/Slic3r #5038: Slic3r Crashes

scuttle130, 2020-12-30:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Slic3r crashes on Mac when right click on model and then select "Settings"  Also crashes from the Menu bar when selecting "Settings."

**To Reproduce**
Steps to reproduce the behavior:
1. Add Model
2. Right click on model
3. Select "Settings" (like to pick extruder)
4. See Slic3r close unexpectedly.

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
Should bring up the normal setting window so you pick extruder, etc.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 10.15.7.
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Slic3r was working just fine.  Then i added a model (3d cube to be exact).  I used the cut feature from the tool bar and cut the model in half via "Z axis", keeping the lower half.  This worked fine.  The next time i tried to select "Settings" on the model, the program quit unexpectedly.  That is the only i have done different in the past 30 times of using the software.
[Cube.stl.zip](https://github.com/slic3r/Slic3r/files/5755775/Cube.stl.zip)
[config.ini.zip](https://github.com/slic3r/Slic3r/files/5755780/config.ini.zip)

lordofhyphens, 2021-03-15:
Can't reproduce on Linux with current development.

JamesDC16, 2022-01-23:
I just got the same problem. It was working fine up to today. I am on macOS Monterey version 12.1. When I add a STL and select Preview, it quit halfway during the processing. I tried the same steps as this post (changing the extruder) and it quits instantly. In the report, the process crashing is perl-local [1108]. I have no idea what it means. Please help