slic3r/Slic3r #5040: Missing top layers

u2d9c9117, 2021-01-07:
**Describe the bug**
Top layers are sometimes missing but the job file is complete, i.e. the gcode includes the end code for return to X0, cooldown etc.

**To Reproduce**
Reproduction has proven difficult for me. I have included a link below for a model I have seen this with though. The settings below resulted in about 10mm missing from the top of the model from the print. Other settings (I have a vague recollection of having tried with a different support material pattern) result in more layers missing (in preview, not tried printing).

Originally I thought this was a render preview issue but for testing purposes I printed the model below with the settings below and the top layers were actually missing from the print. The print head returned to X0 and cooled down as one would expect from a complete model. I can rule out an incomplete gcode file due to transfer failure etc, it did contain end gcode and settings comments.

**3D Model and Slic3r Configuration Export**
I found this with some slicing of Satsana_4010_Version.stl of https://www.thingiverse.com/thing:4647053 

**Expected behavior**
All layers present

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Debian (testing)
 - Version 1.3.0

**Additional context**
; avoid_crossing_perimeters = 1
; bed_shape = 0x0,220x0,220x220,0x220
; bed_temperature = 50
; before_layer_gcode = 
; between_objects_gcode = 
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_connections_width = 0
; brim_width = 5
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 3
; duplicate_distance = 6
; end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
; end_gcode = M104 S90 ; set extruder temperature\nM140 S35; set heatbed temperature \n\nG28 X0  ; home X axis\nM84    ; disable motors\n\nM106 S255 ; fan at full speed\nM109 R90 ; wait for extruder temperature\nM190 R35; wait for  heatbed temperature \n\nM107 ; turn off fan\nM104 S0 ; turn off extruder heating\nM140 S0 ; turn off bed heating\n
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 1
; fan_below_layer_time = 60
; filament_colour = #5E5C64
; filament_cost = 33
; filament_density = 1.24
; filament_diameter = 1.75
; filament_max_volumetric_speed = 0
; filament_notes = ""
; first_layer_acceleration = 0
; first_layer_bed_temperature = 50
; first_layer_extrusion_width = 200%
; first_layer_speed = 20
; first_layer_temperature = 210
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; has_heatbed = 1
; infill_acceleration = 0
; infill_first = 0
; interior_brim_width = 0
; layer_gcode = 
; max_fan_speed = 100
; max_layer_height = 0.3
; max_print_speed = 80
; max_volumetric_speed = 0
; min_fan_speed = 100
; min_layer_height = 0.15
; min_print_speed = 10
; min_skirt_length = 5
; notes = 
; nozzle_diameter = 0.4
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 0
; printer_notes = 
; resolution = 0
; retract_before_travel = 2
; retract_layer_change = 1
; retract_length = 1
; retract_length_toolchange = 10
; retract_lift = 0
; retract_lift_above = 0
; retract_lift_below = 0
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 60
; skirt_distance = 6
; skirt_height = 1
; skirts = 5
; slowdown_below_layer_time = 5
; spiral_vase = 0
; standby_temperature_delta = -5
; start_filament_gcode = "; Filament gcode\n"
; start_gcode = G21\nG90\nM82\n\nM104 S[first_layer_temperature] ; set extruder temp\nM140 S[first_layer_bed_temperature] ; set bed temp\nG28 ; home all axes\nG0 X0 Y180 Z1 F9000 ; move to line beginning raise nozzle but not too much allowing blob to land on bed\nM190 S[first_layer_bed_temperature] ; wait for bed temp\nM109 S[first_layer_temperature] ; wait for extruder temp\n
; temperature = 205
; threads = 12
; toolchange_gcode = 
; travel_speed = 150
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_set_and_wait_bed = 0
; use_set_and_wait_extruder = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 1
; z_offset = 0
; z_steps_per_mm = 0
; adaptive_slicing = 0
; adaptive_slicing_quality = 75%
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.25
; infill_only_where_needed = 1
; interface_shells = 0
; layer_height = 0.2
; match_horizontal_surfaces = 1
; raft_layers = 0
; regions_overlap = 0
; seam_position = aligned
; sequential_print_priority = 0
; support_material = 1
; support_material_angle = 0
; support_material_buildplate_only = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_extrusion_width = 0
; support_material_interface_layers = 3
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_max_layers = 0
; support_material_pattern = pillars
; support_material_spacing = 2.5
; support_material_speed = 60
; support_material_threshold = 55%
; xy_size_compensation = 0
; bottom_infill_pattern = rectilinear
; bottom_solid_layers = 4
; bridge_flow_ratio = 1
; bridge_speed = 30
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 15%
; fill_gaps = 1
; fill_pattern = 3dhoneycomb
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 60%
; infill_speed = 60
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 50
; perimeters = 4
; small_perimeter_speed = 20
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 20
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_infill_pattern = rectilinear
; top_solid_infill_speed = 15
; top_solid_layers = 4

lordofhyphens, 2021-03-15:
I loaded your config and the model. What is broken here? I can't seem to reproduce.

[image]

u2d9c9117, 2021-03-16:
Thanks for taking your time to look into it.

That's not what it looked like for me though I'm not 100% sure whether I might have positioned it slightly differently on the base. Basically the top layers were missing in the 3D rendering and the printout but the gcode file appeared to be complete (including final park position, cooldown etc). It had not happened before nor has it since. Feel free to close if you wish. If it happens again I'll try to capture more data (full gcode file, screenshots, photos, etc.).

lordofhyphens, 2021-03-16:
I loaded your provided config and imported the referenced model. If there
was rotation, etc, that needs to be done to get it to break, I recommend
exporting the plate once you get it to break, archiving in a ZIP and then
dropping on the issue.

lordofhyphens, 2021-03-21:
Closing, can't reproduce.

closed 2021-03-21 by lordofhyphens