slic3r/Slic3r #5041: Slic3r 1.3.0.64bit won't start on Windows 10 x64, perl524.dll missing error

ShayneP42, 2021-01-12:
I downloaded Slic3r-1.3.0.64bit.zip today (2021.01.12) and unzipped with Windows included "Extract All". I copied the resulting "Slic3r-1.3.0.64bit" folder to my C:\Program Files folder. Clicked the "yes, I'm an Admin". Then double clicked the slic3r.exe file, and get this error:
The code execution cannot proceed because perl524.dll was not found. Reinstalling the program may fix this problem. 

I didn't "install" it per se. So.......... ????? I tried it in Win7 and Win8 compatibility modes. Tried with antivirus software off. Tried it as Admin.  Other thoughts?

[image]

lordofhyphens, 2021-03-15:
Can't reproduce on a Windows 10 virtual machine with a fresh checkout.

lordofhyphens, 2021-03-15:
https://dl.slic3r.org/dev/win/Slic3r-master.2021.03.15.2475.5d371aab.64bit-PR.zip 

Closing, as I can't reproduce and the file is in current output.

closed 2021-03-15 by lordofhyphens