slic3r/Slic3r #5042: Remove unnecessary null pointer checks

elfring, 2021-01-17:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [AvoidCrossingPerimeters](https://github.com/slic3r/Slic3r/blob/92abbc42dfdd5385c1f9c3a450e2f3da835f8b8d/xs/src/libslic3r/GCode.cpp#L17 "Destructor for AvoidCrossingPerimeters")
- [ConfigOptionDef](https://github.com/slic3r/Slic3r/blob/92abbc42dfdd5385c1f9c3a450e2f3da835f8b8d/xs/src/libslic3r/ConfigBase.cpp#L208 "Destructor for ConfigOptionDef")
- [Option](https://github.com/slic3r/Slic3r/blob/92abbc42dfdd5385c1f9c3a450e2f3da835f8b8d/src/GUI/OptionsGroup.hpp#L34 "Destructor for Option")

roy-stewart, 2021-01-24:
I second this. It won't make a functional difference but it will definitely clean up the code base. I'm working on a pull request to remove these null pointer checks now.

lordofhyphens, 2021-01-24:
@roy-stewart I will review it.

elfring, 2021-01-24:
How do you think about to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

lordofhyphens, 2021-02-13:
Merged, thank you.

I currently don't want to deal with the diff from clang-tidy or agonize over the ruleset.

closed 2021-02-13 by lordofhyphens