slic3r/Slic3r #5045: Repetier Server added to upload server list

jmger, 2021-01-25:
I am using Repetier Server. I see only Octoprint and Duet as server options. Can Repetier Server be added to the list?

### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lordofhyphens, 2021-01-25:
I don't user Repetier Server and have zero interest in attempting to support closed-source software. I *might* accept a Pull Request to add the support, but my first recourse to Repetier changing something and breaking Slic3r is to disable the feature,

Siox0911, 2021-02-13:
To do a break with both worlds. It is not necessary that Slic3r is support Repetier-Server. You can use Repetier-Host (written in C#). Repetier-Host is supporting Slic3r. Also you can add other slicers too. I like this tool. One UI multible slicers. You can switch between without new model placing. 

@lordofhyphens Sorry, but the latest stable version of Slic3r is 6 years old. Version 1.3.0 is 3 years old. I think this project is gone.

lordofhyphens, 2021-02-13:
@Siox0911 "latest stable" is 1.3.0.

Siox0911, 2021-02-13:
@lordofhyphens 
[image]

jmger, 2021-02-13:
I have since went back to using Simplify3D. Nothing is perfect. Simplify3D hardly ever puts out updates. But for the most part things just work. I find with open source software that a lot of the time there is just to many bugs for me. Take Marlin printer firmware for one example. I tried so many different version trying to find one version than would even compile without errors. The errors were not on my end. I try compiling the code straight off with out any modification to see if everything works. Before I try to edit the software/firmware. I had to stick to the Tevo Tornado v1.1. So if I don't mind if things brake then Open source is great! But if I need, need ,need things to just work then commercial software is usually better. There are exception of course.
I love repetier Server. It was very easy to install. I did have a few times recently that Repetier Server went into a boot loop and I had to reinstall the software on my Raspberry Pi. But that was after trying to upload a gcode file or 2 from Slic3r. I have not had any issues with Repetier Server uploading files from Simplify3D. But then you have software like Lumafusion & Shapr3D for the ipad where they are adding new stuff all the time. And very responsive on their respective forums.

lordofhyphens, 2021-02-13:
@jmger They're also ostensibly getting paid. I do this in my free time (of which I have precious little). Prusa's people did their own fork (and again, are paid to work on it). 

Most users of FOSS 3d printing software want stuff done (features added, etc) but can't or won't actually contribute.

lordofhyphens, 2021-02-13:
@alranel https://app.circleci.com/pipelines/github/slic3r/website/40/workflows/19e67aa5-92a7-46f0-ab91-1b88235572f5/jobs/43 

CircleCI broke trying to pull down code.

jmger, 2021-02-14:
You are correct but that is the advantages and disadvantages of both types
of software. I like the concept of open source software. I like thast my
Tevo Tornado printer is so easy to upgrade. Unlike my first printer the XYZ
Devinci  1.0 printer and my Monoprice Mini Delta. Both printers are really
not very upgradable and hard to fix. I just for 3D printing need something
that works. At least on the Marlin firmware side. I don't need cutting edge
features. I just need it to work.