slic3r/Slic3r #5046: Auto orientation option

jmger, 2021-01-25:
Can there be some sort of option to better/easier aline an object on the Bed surface. I know this is an issue with Simplify3D as well. I will give you an example: I use Shapr3D on my iPad Pro to design my parts. One part is a simple cap design. A flat 3" surface on the top layers. I would like to print the design on the top of the cap as the first layer. So flat on the bed. It is very hard to line this up. I wish I could tell Slic3r that I want to orient the part so it will print flat on the build surface. Right now even though I think the part is laying flat it appears that it must not be completely flat when I check it in the Adaptive layer section. At least Simplify3D, besides entering a specific value, allows you to press an arrow up/down button to move the part in any direction. At least this is a little easier than just entering values. Also simplify3D has a Drop on bed option. This is useful for me to know that the software is placing the full part on the bed and not below the bed surface. 

Thank You

### Do not request features for the Prusa3D fork of Slic3r here without confirming it is applicable to the mainline Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

dwillmore, 2021-01-25:
There is the 'rotate surface to bed' function which sounds like it would do
exactly what you want.

lordofhyphens, 2021-03-15:
Right-click on the model in the plater.
[image]
Select the face you want to be on the bed.
[image]

closed 2021-03-15 by lordofhyphens