slic3r/Slic3r #5054: Slic3r 1.2.9 crashes under Linux - 1.3.1 fixes it but needs to be pushed out to the repositories.

SteveBaker, 2021-02-28:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

lordofhyphens, 2021-03-15:
I don't control what distros do.

SteveBaker, 2021-03-15:
Generally, when a distro decides to pick up a software package to
include, they give you a contact email for the person who controls the
distribution of your package so you can inform them of problems like
this. 

That's really necessary because the sudden distribution of code that
breaks a package is a serious matter for people who auto-update their
machines.

--- 

 

Links:
------
[1] https://github.com/slic3r/Slic3r/issues/5054#issuecomment-799038761
[2]
https://github.com/