slic3r/Slic3r #5055: wont compile with boost 1.75.0

ferdna, 2021-03-06:
Attempting to create directory /tmp/SBo/Slic3r-1.3.0/local-lib
App::cpanminus is up to date. (1.7044)
--> Working on Class::XSAccessor
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/Class-XSAccessor-1.19.tar.gz ... OK
Configuring Class-XSAccessor-1.19 ... OK
Building and testing Class-XSAccessor-1.19 ... OK
Successfully installed Class-XSAccessor-1.19
1 distribution installed
Devel::CheckLib is up to date. (1.14)
--> Working on Encode
Fetching http://www.cpan.org/authors/id/D/DA/DANKOGAI/Encode-3.08.tar.gz ... OK
Configuring Encode-3.08 ... OK
Building and testing Encode-3.08 ... OK
Successfully installed Encode-3.08 (upgraded from 3.06)
1 distribution installed
--> Working on Encode::Locale
Fetching http://www.cpan.org/authors/id/G/GA/GAAS/Encode-Locale-1.05.tar.gz ... OK
Configuring Encode-Locale-1.05 ... OK
Building and testing Encode-Locale-1.05 ... OK
Successfully installed Encode-Locale-1.05
1 distribution installed
--> Working on ExtUtils::CppGuess
Fetching http://www.cpan.org/authors/id/E/ET/ETJ/ExtUtils-CppGuess-0.21.tar.gz ... OK
Configuring ExtUtils-CppGuess-0.21 ... OK
==> Found dependencies: Module::Build, Capture::Tiny
--> Working on Module::Build
Fetching http://www.cpan.org/authors/id/L/LE/LEONT/Module-Build-0.4231.tar.gz ... OK
Configuring Module-Build-0.4231 ... OK
Building and testing Module-Build-0.4231 ... OK
Successfully installed Module-Build-0.4231
--> Working on Capture::Tiny
Fetching http://www.cpan.org/authors/id/D/DA/DAGOLDEN/Capture-Tiny-0.48.tar.gz ... OK
Configuring Capture-Tiny-0.48 ... OK
Building and testing Capture-Tiny-0.48 ... OK
Successfully installed Capture-Tiny-0.48
Building and testing ExtUtils-CppGuess-0.21 ... OK
Successfully installed ExtUtils-CppGuess-0.21
3 distributions installed
--> Working on ExtUtils::MakeMaker
Fetching http://www.cpan.org/authors/id/B/BI/BINGOS/ExtUtils-MakeMaker-7.60.tar.gz ... OK
Configuring ExtUtils-MakeMaker-7.60 ... OK
Building and testing ExtUtils-MakeMaker-7.60 ... OK
Successfully installed ExtUtils-MakeMaker-7.60 (upgraded from 7.44)
1 distribution installed
--> Working on ExtUtils::ParseXS
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/ExtUtils-ParseXS-3.35.tar.gz ... OK
Configuring ExtUtils-ParseXS-3.35 ... OK
Building and testing ExtUtils-ParseXS-3.35 ... OK
Successfully installed ExtUtils-ParseXS-3.35 (downgraded from 3.40)
1 distribution installed
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.78)
--> Working on Getopt::Long
Fetching http://www.cpan.org/authors/id/J/JV/JV/Getopt-Long-2.52.tar.gz ... OK
Configuring Getopt-Long-2.52 ... OK
Building and testing Getopt-Long-2.52 ... OK
Successfully installed Getopt-Long-2.52 (upgraded from 2.51)
1 distribution installed
--> Working on IO::Scalar
Fetching http://www.cpan.org/authors/id/C/CA/CAPOEIRAB/IO-Stringy-2.113.tar.gz ... OK
Configuring IO-Stringy-2.113 ... OK
Building IO-Stringy-2.113 ... OK
Successfully installed IO-Stringy-2.113
1 distribution installed
--> Working on Module::Build::WithXSpp
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/Module-Build-WithXSpp-0.14.tar.gz ... OK
Configuring Module-Build-WithXSpp-0.14 ... OK
==> Found dependencies: ExtUtils::XSpp
--> Working on ExtUtils::XSpp
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/ExtUtils-XSpp-0.18.tar.gz ... OK
[... 266 lines cut ...]
                 from src/libslic3r/SLAPrint.cpp:1:
/usr/include/boost/bind/placeholders.hpp:58:38: note:   ‘boost::placeholders::_1’
   58 | BOOST_STATIC_CONSTEXPR boost::arg<1> _1;
      |                                      ^~
In file included from /usr/include/boost/mpl/aux_/include_preprocessed.hpp:37,
                 from /usr/include/boost/mpl/placeholders.hpp:43,
                 from /usr/include/boost/iterator/iterator_categories.hpp:16,
                 from /usr/include/boost/iterator/advance.hpp:11,
                 from /usr/include/boost/next_prior.hpp:25,
                 from /usr/include/boost/thread/futures/wait_for_any.hpp:20,
                 from /usr/include/boost/thread/future.hpp:41,
                 from /usr/include/boost/thread.hpp:24,
                 from src/libslic3r/libslic3r.h:17,
                 from src/libslic3r/SLAPrint.hpp:4,
                 from src/libslic3r/SLAPrint.cpp:1:
/usr/include/boost/mpl/aux_/preprocessed/gcc/placeholders.hpp:29:16: note:   ‘mpl_::_1’
   29 | typedef arg<1> _1;
      |                ^~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-undefined-var-template’ may have been intended to silence earlier diagnostics
error building src/libslic3r/SLAPrint.o from 'src/libslic3r/SLAPrint.cpp' at /usr/share/perl5/ExtUtils/CBuilder/Base.pm line 185.
! Installing ./xs failed. See /root/.cpanm/work/1615004113.26106/build.log for details. Retry with --force to force install it.
FAIL
1 distribution installed
The XS/C++ code failed to compile, aborting

lordofhyphens, 2021-03-14:
@ferdna Try it now, I merged a PR having to deal with `_1`.

ferdna, 2021-03-15:
> @ferdna Try it now, I merged a PR having to deal with `_1`.

it works now. thank you so much. :)

closed 2021-03-15 by lordofhyphens