slic3r/Slic3r #5057: Linux development builds are empty files

msrd0, 2021-03-06:
The development builds for linux all have a size of 0 bytes: https://dl.slic3r.org/dev/linux/. That site is linked to from your download page (https://slic3r.org/download/). The other operating systems have some content but they probably won't run on Linux.

lordofhyphens, 2021-03-06:
Thanks, I need to do some cleanup on the file storage.

Bluelinegecko, 2021-03-13:
The latest windows dev. build files (2-13-2021)  are empty too and have been for a month. The previous dev release  listed on the download page is from August of 2020 and has several bugs.

lordofhyphens, 2021-03-13:
Should be sorted out now

msrd0, 2021-03-13:
Yes, it's working again. Thanks

closed 2021-03-13 by msrd0

Bluelinegecko, 2021-03-14:
The windows files are still broken. The newest 64-bit release fails to complete. Tried on multiple computers and browsers. I'm about to ditch slic3r permanently for Prusaslicer as the state of development over the last 1-2 years has just not been what it once was.

lordofhyphens, 2021-03-14:
@Bluelinegecko I literally just downloaded the 64-bit release and opened it, and it started right up. From https://dl.slic3r.org/dev/win/Slic3r-master.2021.03.13.2462.0536b14f.64bit-PR.zip

----

@Bluelinegecko for as long as I have been associated with Slic3r, there has been 1-2 people at most actively working on code (in their spare time), and hundreds/thousands that ask for stuff. If you want things to get better, *come and help*. Pay other people to do development work, design tests, etc. 

Prusa's people are paid to work on their fork (and not to contribute any of it back up).

Bluelinegecko, 2021-03-15:
@lordofhyphens I deleted my previous comment after I realized that the link did work. For some reason my antivirus program was taking forever (several minutes) to complete the automatic scan of the latest file download, whereas it would scan the previous file version in seconds. I'm not sure why.

I understand the Slic3r team is vastly understaffed, all volunteer, and could use more support. I am not a programmer by trade or trust me I would help. I have been a big fan of slic3r since I got into 3d printing years ago and it has been my go to slicer since almost the beginning. You guys did a great job for years. BUT.... In the last 1-2 years it seems the pace and quality of development has been abysmal. It has been almost 2 years since an official release. The automated development links were broken for several weeks or possibly months. 3 year old issues like #4385 still are a problem in newest development version and make using Slic3r daily a real pain. Very few features have been added in the last 2 years, and features such as the "rotate model 45°" buttons have even been inexplicably removed. It is fair for you to say "Put up some money and help" and it is equally fair for me to ask why I would bother investing anything in something that is showing so little life or progress. It is a catch-22 situation for sure. I have an overdeveloped sense of loyalty which is probably why I have stuck to Slic3r for so long but honestly I think Slic3r is so far behind Cura and Prusaslicer that it is time to rethink the path forward. It is hard not to feel like Slic3r has been a slowly sinking ship these last 2 years. 

For what it is worth here is my suggestion. Since Prusaslicer is opensource as well, make the next version of Slic3r a brand new fork of Prusaslicer;s current code. Let the project come full circle. Don't bother wasting untold hours time replicating/porting their work as they have already squashed a ton of bugs and added features slic3r doesn't have. Instead focus on removing the commercialization they added and removing the tie-ins to their own products, and make it a better version of a universal slicer like Slic3r originally was. Then build from there. The only thing I see in Slic3r that isn't in Prusaslicer yet (and should be) is the "Z full steps/mm" setting which is helpful especially when using adaptive layers. I'm sure there might be another feature or two unique to slic3r, but Prusaslicer's better looking and more responsive UI, and additional features make it the clear winner right now. I'm not a programmer as I wrote earlier so maybe my suggestion isn't as easy or feasible as I think, but my patience with Slic3r is worn out and for the last year or so I have not been able to recommend Slic3r to other 3d printers. I wont be able to ever again until it can catchup to Cura or Prusaslicer in responsiveness and features. If Slic3r starts showing it is on par with it's peers again and has something to set it apart form the others maybe it will start attracting some more investment from the community, but if things stay as they currently are I'm sure history will record that Slic3r effectively died in 2018 with version 1.3. I suspect a lot of former fans of slic3r feel the same way.

lordofhyphens, 2021-03-16:
Something that would provide a lot of value and doesn't require someone being a programmer would be to comb through the issue tracker, retest stuff that may be fixed spontaneously, and opine as to how the hundreds of one-off feature requests might get done, how they would work from a user's POV that isn't magic.