slic3r/Slic3r #5059: Thread 2 terminated abnormally crash

Tyql, 2021-03-06:
**Describe the bug**
Slic3r crashes when I'm scaling the object

**To Reproduce**
Steps to reproduce the behavior:
1. Add .STL
2. Select object
3. Scale > Z Axis any %
4. Scale > Y or X Axis any %
4. Program crashes.

**3D Model and Slic3r Configuration Export**
[Bug Report.zip](https://github.com/slic3r/Slic3r/files/6095873/Bug.Report.zip)

**Expected behavior**
Not crashing

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 64 Bit
 - Version: 1.3.0

lordofhyphens, 2021-03-15:
Well, it isn't crashing on Linux, so there's that.

lordofhyphens, 2021-03-15:
Didn't crash either with the build for 0536b14f.

https://dl.slic3r.org/dev/win/

lordofhyphens, 2021-03-15:
I need some more information about what you were doing here; as loading the provided configuration and model and doing what you said did not cause a crash on current dev version. 

Closing, will re-open if you can reproduce on current -dev.

closed 2021-03-15 by lordofhyphens