slic3r/Slic3r #5060: Gcode to heat second hotend not generated when ooze prevention off.

perrito666, 2021-03-06:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
When printing with double hotend with `ooze_prevention=false` `T1` is never heated, in my case I see `M104 S205 T0 ; set temperature` appear in the right place but when the turn is for `T1` to print there is no `M104`, this causes extruder to push filament on a cold nozzle.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure 2 extruders (no shared nozzle)
2. Disable ooze prevention
3. Slice a multi material print
4. Resulting GCode should have only a `M104` for `T0`

**3D Model and Slic3r Configuration Export**
[slic3r_bug_data.zip](https://github.com/slic3r/Slic3r/files/6096014/slic3r_bug_data.zip)

**Expected behavior**
This should be generated before tool Use (for a sample case of 205C)
```gcode
M104 S205 T1
M109 S205 T1
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.55.2

**Additional context**
Add any other context about the problem here.

dwillmore, 2021-03-06:
That's not slic3r, that's Prusa slic3r which is not maintained here.
Please see the link at the top of your bug report where it directs you to
the Prusa slic3r issue tracker.

closed 2021-03-06 by dwillmore

perrito666, 2021-03-06:
Apologies 🤦 I missunderstood that prusa slicer was a fork of your project and not the other way around.

lordofhyphens, 2021-03-07:
PrusaSlicer is a fork of Slic3r.

If you can reproduce with Slic3r 1.3.0, this can be reopened.