slic3r/Slic3r #5066: Unexpected solid layers on slopes

dwillmore, 2021-03-15:
I am seeing extra solid draws under and over slopes.  It appears to be dependant on the angle of the slope. 
[Bug_Report_1.zip](https://github.com/slic3r/Slic3r/files/6142664/Bug_Report_1.zip)

Steps to reproduce the behavior:
Load the included model and config.  Preview the slice and go to the layer view.  Use the slider to view somewhere around the middle of the model.  Note the extra solid draw near the sloped walls.  This is unexpected.

[image]

 - OS: Linux, Fedora 32
 -  - Version Slic3r-1.3.1-dev-c61c25b-x86_64.AppImage

dwillmore, 2021-03-15:
Does not seem to depend on the infill value.  Tried 1%, 5% (shown), 15%, and 50% and noticed no variation.

lordofhyphens, 2021-03-15:
https://github.com/prusa3d/PrusaSlicer/issues/1054

lordofhyphens, 2021-03-15:
https://github.com/prusa3d/PrusaSlicer/issues/932
https://github.com/supermerill/SuperSlicer/issues/253

lordofhyphens, 2021-03-15:
https://github.com/slic3r/Slic3r/issues/4496

dwillmore, 2021-03-15:
https://user-images.githubusercontent.com/10374140/111192555-91ed5600-858f-11eb-97e4-795aee3443a7.mp4

I produced an animation from a series of screenshots showing the bottom few layers.  It starts just above the last solid bottom layer and goes until it reaches 'steady state'.