slic3r/Slic3r #5067: Unable to open AM4 file

faeluke, 2021-03-20:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)
I would never!

**Describe the bug**
Sic3r does not open a specific set of AM4 files that I needed to print. Files open in Cura, If I save them in Cura as STL files they open fine. 
 
**To Reproduce**
Steps to reproduce the behavior:
Open the file provided in the zip file attached

**3D Model and Slic3r Configuration Export**
Not required, since the file doesn't open

**Expected behavior**
Open the file

**Screenshots**
[image]

**Desktop (please complete the following information):**
MacOS 10.15.

**Additional context**
Using development release as the version supporting gyroid infill is not officially released

[01-Spindle-Mount-65mm-2a.amf.zip](https://github.com/slic3r/Slic3r/files/6175435/01-Spindle-Mount-65mm-2a.amf.zip)

lordofhyphens, 2021-03-20:
Is it a compressed AMF file?

faeluke, 2021-03-20:
I have no idea additional knowledge about the file. it can be opened in Cura. I have a workaround but was wondering if Slic3r should handle it directly? I can add it as a feature request instead. The extension was linked to Slic3r. I have attached the resultant STL file that Cura outputs. 
Also note that when the file opens from double-clicking it, a new instance of Slic3r opens and does not display any error message. 

[01-Spindle-Mount-65mm-2a.stl.zip](https://github.com/slic3r/Slic3r/files/6175817/01-Spindle-Mount-65mm-2a.stl.zip)

One of two things (or both) should happen, an error message should be displayed if a file of this type 
is opened, and or opening this type of file is added as a feature. 
Thanks for reaching out, I am sorry that I don't have any more data to share. I have more files like this if you need them. 
Luke

lordofhyphens, 2021-03-20:
Edit it in a text editor. If it looks like garbage, it's compressed.

faeluke, 2021-03-20:
You are correct, it is compressed. That doesn't mean Slic3r should open and do nothing. Sorry, my incomplete understanding of the issue led me to report the wrong problem. 

Now that we both have a better understanding of the issue:

Double-clicking a file with a .AM4 extension on my machine opens Slic3r
The file is opened
The data is displayed
 
Double-clicking this file opens Slic3r
Slic3r attempts to open the file
Nothing happens 

When I use the open file dialog box inside Slic3r an error message occurs. 

In the least, I expect the same invalid file dialog box to open and tell that the normal process occurred, but something was wrong with the file. 

Thanks, 

I will put in a feature request for Slic3r to process compressed files as Cura does.

lordofhyphens, 2021-03-20:
There is a PR open to support deflated AMF files.

lordofhyphens, 2021-03-20:
Duplicate of #4511

lordofhyphens, 2021-03-20:
@faeluke  https://dl.slic3r.org/dev/mac/jenkins/slic3r-deflate-amf-18ddd3a3-osx-deflate-amf.dmg give that a try.

lordofhyphens, 2021-03-20:
@faeluke as a workaround, decompress with 7zip or something and it should read.

faeluke, 2021-03-20:
Thanks,

I actually renamed the file to .zip and used the built-in utility on the
Mac before I replied to your earlier post.

It didn’t occur to me to do that until you mentioned it was most likely a
compressed file.

My other work around was to open the 12 files in CURA and save them as STL.

faeluke, 2021-03-20:
> @faeluke <https://github.com/faeluke> https://dl.slic3r.org/dev/mac/
> jenkins/slic3r-deflate-amf-18ddd3a3-osx-deflate-amf.dmg give that a try.

Will do, down another rabbit hole at the moment.

faeluke, 2021-03-21:
This had no effect on the issue.

lordofhyphens, 2021-03-21:
@faeluke  Should load now after jenkins gets done building the PR. 

Whenever this is live (expect ~20-30 minutes from this post). I had had an issue where I couldn't actually start the GUI and for some reason my tests were passing (which I'm looking into now); assuming I'm didn't actually hallucinate that.
https://dl.slic3r.org/dev/mac/jenkins/slic3r-deflate-amf-69e27ec20-osx-deflate-amf.dmg

lordofhyphens, 2021-03-21:
Apparently I hadn't fully added my tests from my old code. So now they pass _and_ it works from the GUI.

faeluke, 2021-03-21:
The link is not active, will look again in the AM. 

Thanks for looking at it!!

closed 2021-03-21 by lordofhyphens

faeluke, 2021-03-21:
Not convinced you nailed it yet... 

I just tested the release below and the file does not open.
slic3r-quick-port-amf-18ddd3a36-osx-quick-port-amf.dmg

lordofhyphens, 2021-03-21:
@faeluke https://dl.slic3r.org/dev/mac/jenkins/slic3r-1.3.1-dev-8c6927845-osx.dmg

faeluke, 2021-03-21:
It works, thanks!

faeluke, 2022-11-19:
I just tried to redownload this to my new mac, and part of the slic3r site
is down. I can move to another slicer, but I really appreciate the features
and simplicity of Slic3r.

Thanks,

Luke

PR #5068 (2021-03-21): Quick port amf
Fixes #4511 
Fixes #5067