slic3r/Slic3r #5071: Nozzle diameter not recognized

xflowx, 2021-04-01:
**Describe the bug**
I'm using Slic3r 1.3.0 on Ubuntu 20.04 in RepetierHost and Standalone. Whatever nozzle diameter I input in the "Printer settings", the value used is always equal the layer height from "Print settings"

**To Reproduce**
Set Layer height to 0.3
Set Nozzle Diameter to anything else than 0.3
Slice
Check gcode file header, lines 3-7: extrusion widths will be based on Layer height, not nozzle diameter

**3D Model and Slic3r Configuration Export**
Behavior occurs with all 3D models
attached is the configuration where i have found the issue, gcode was generated with this exact settings

**Expected behavior**
Extrusion width is based on nozzle diameter

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version 1.3.0
    Happens with AppImage as well as installation from package sources, and also if used out of RepetierHost

[files.zip](https://github.com/slic3r/Slic3r/files/6242323/files.zip)

lordofhyphens, 2021-04-01:
Nozzle diameter is required to be  > layer height

xflowx, 2021-04-01:
> Nozzle diameter is required to be > layer height
> […](#)

It is. In the uploaded example, layer height is 0.3, nozzle diameter is 0.8

xflowx, 2021-04-01:
actually, the previous comment made me think, and i tried one more thing: 
Layer Height is set to 0.3mm
nozzle diameter = 0.8 => used extrusion width = 0.3
nozzle diameter = 0.2 => used extrusion width = 0.2

So, it seems that in reality "Nozzle diameter is required to be  **<** layer height" which obviously makes no sense...

lordofhyphens, 2021-04-01:
Nozzle diameter is the upper bound on layer height (and Slic3r will yell at
you if you try to exceed it) and the practical lower bound on line width. I
don't think I have seen something like this before but I can't look at it
now

kanethemediocre, 2021-04-05:
I experienced this as well, in windows 7.  I always use 1mm nozzle, and 0.5mm layers, and slic3r kept ignoring my nozzle size and instead used (at least approximately) 0.5mm paths.  I never correlated this with the layer height or checked the g-code like the OP did, I just gave up slic3r for cura over this problem, but I hate cura and wish I could use slic3r.

xflowx, 2021-04-06:
> Nozzle diameter is the upper bound on layer height (and Slic3r will yell at you if you try to exceed it) and the practical lower bound on line width. I don't think I have seen something like this before but I can't look at it now
> […](#)

Indeed, i would also think of it as you explained, therefore i've reported;-) let me know if you need some more info, i'm following this thread.