slic3r/Slic3r #5076: Vulnerable copy of libexpat in folder /xs/src/expat?

hartwork, 2021-04-26:
Hi!

From a quick look, the copy of libexpat at https://github.com/slic3r/Slic3r/tree/master/xs/src/expat is old and vulnerable — every release before 2.2.8 is. Are you aware?

Best, Sebastian

hartwork, 2021-06-14:
Any thoughts?

supermerill, 2021-06-14:
You can create an amf virus?
I'll see if using a newer version works.

hartwork, 2021-06-14:
> You can create an amf virus?

I'm not sure what you refer to here. What do you mean?

> I'll see if using a newer version works.

Cool!

supermerill, 2021-06-14:
> I'm not sure what you refer to here. What do you mean?

you said that libexpat contains vulnerabilities (I don't know what kind). As it's used for amf & 3mf reading/writing, the way these vulnerabilities can affect slic3r is maybe by creating an amf or .3mf virus ?

hartwork, 2021-06-14:
@supermerill I see, thanks for elaborating. I wouldn't call it a virus myself, but it would be possible to craft a file that will have undesired affects when opened, e.g. causing use of so much RAM that you will need to reboot the the machine and lose any unsaved work at that moment (referring to Billion Laughs Attacks). Given how old this copy is, there are more attack vectors possible. Please see the change log at https://github.com/libexpat/libexpat/blob/master/expat/Changes for more details.

hartwork, 2024-11-13:
Closing to keep my https://github.com/issues clean…

closed 2024-11-13 by hartwork