slic3r/Slic3r #5085: Feature Request: Perimeters/solid lines before gap fill

LightningShark25, 2021-05-31:
I'd like to request a feature to ensure gap fill > certain area, is performed AFTER perimeter/solid lines are laid down on the first layer

The issue comes up for me when printing models with high detail, and potentially a lot of gap fill (obvously depending on nozzle size etc)
The current behavior seems to be that gap fill is done at the same time (often before) perimeters are put down.  The issue is that sometime this fine detail can be as small as a single dot, which doesn't adhere well to the bed.   On my printer, the printer tries to put down such a small volume of plastic that it doesn't come out of the nozzle well, or adhere to the bed properly, resulting in a messy first layer.  This then affects the rest of the print.

I'd like to see the perimeter / more defined lines > certain area laid down first, which have a higher chance of adhering to the bed properly. Then the printer can come back and fill the gaps in those solid lines with gap fill.  Because it is filling gaps between solid lines, plastic should adhere to existing plastic (and the bed) better than just the bed alone.

I realise that this may increase the print time as it will involve extra x and y movements (and retracts).  Hence why this should be configurable for those users that don't want it.  If not configurable on/off (preferable) then the area (or volume of plastic - less desirable but potentially easier to program) > a certain value, will mean users can set this value significantly high so that the same logic applies, but won't trigger.

This feature is most critical on the first layer.  Ideally this is for the first layer only, so the rest of the print will be quicker.

Happy to discuss or clarify if not clear

Thanks
Kev

supermerill, 2021-05-31:
gapfill should happen only after the perimeters (in a given area)
Do you have a project that show that bad behavior?

LightningShark25, 2021-05-31:
I'm closing this issue as I realised I posted in the wrong thread sorry

closed 2021-05-31 by LightningShark25