slic3r/Slic3r #5094: proposal for different print resolution on the same object

atomaticSrl, 2021-07-29:
Hello,

during the use of slic3r it happens to me, for my working needs, to have the need to have areas of the piece
printed with different resolutions.

An example that often happens to me is that of having a solid cylinder where I have to pass a screw because, in some cases,
I prefer to make the hole with a drill bit, both to have more mechanical strength and also to have greater precision.

Typically the high resolution cylinder around the screw hole should have an outside diameter equal to or slightly larger than the screw head or washer to put under the head,

This is just an example, the number of cases in which there is a need to define zones with different resolutions is infinite (as you all know very well)

in the figure that I am attaching there is an simple example (designed ad hoc) of how I would like to be able to define these different areas.

The piece is made up of 3 different layers, the main part, light blue in color, is the part to be printed at low resolution,
the red part is to be printed in high resolution and a yellow part in which I define a support explicitly.

the proposal is to export three different files in .stl (example_hres.stl, example_lowr.stl and example_supp.stl, for example) from your  cad  as usual
and therefore to be able to include these three files in sequence in slic3r and associate a different print setting for each of the different files.

Slic3r must be able to internally generate the 3 (or more) different gcodes and then merge these files to obtain the final single file (defining an appropriate priority level).

(I do not know the internal structure of slic3r, that the merge must take place at the gcode level is just an example that serves to make my proposal understood)

grossly I think that 75% of the code needed for this proposal is already present in the current sw, there is to implement the code for the merge between the different files

As already mentioned, the mechanism can also be useful for defining supports if those automatically generated are not adequate

I hope this proposal will be accepted because it will be very useful to me.
thank you

[image]

SG
-- 
//**************************
//* Giuseppe Saraceno
//*
//* Atomatic Systems srl
//*
//**************************

trysten, 2021-08-12:
If I understand your issue, I suggest your preprocess your models to simplify the geometry.

"Per-Object print settings" feature doesn't seem very useful compared to the work to implement it. I can just print parts one at a time. But if it would be useful to you, try to think of good reasons it would be useful to everyone. Maybe it would be useful for dual-extrusion? 

P.S. I wonder if you know about the layer height table in object settings. If layer height is what you mean by resolution, you may already be able to do what you want. Select your object, click print settings -> Layer Height tab. This allows for per-object layer-height. 

If this sufficiently resolves your need, please close the issue. It will still be open for further comments.

atomaticSrl, 2021-08-12:
I am amazed that my proposal is so difficult to understand, probably even google translator needs some improvement ...

Perhaps it is better understood if I replace the word "resolution" with the phrase "different infill densities in a single object".

the example object that I have attached cannot be printed one part at a time. the two red cylinders are INCLUDED in the semi-transparent light blue rectangular object displayed to show its internal parts.

lordofhyphens, 2021-08-12:
https://manual.slic3r.org/advanced/modifier-mesh

Isn't this modifier meshes?

atomaticSrl, 2021-08-13:
yes, it would seem just what I need.
I did not know this slic3r function.

I'll try it at the first opportunity, however, my proposal seemed slightly better ...

lordofhyphens, 2021-08-13:
I have zero interest in implementing your request, as I believe that
modifier meshes cover the use case for the majority of cases. I will review
a pull request.

If you really want it done, I strongly suggest commissioning a developer to
work on it (and talk out the UX here to get buy-in) and submit a PR. That
developer to be commissioned is not me.

atomaticSrl, 2021-08-14:
Dear Mr. lordofhyphens ,
if you read my last comment carefully (3 rows)  you will notice that I never asked for my proposal to be implemented, least of all to you, whom I don't know. I just wrote that "it seems to me that my proposal is slightly better".

The curse and the arrogance of his response towards a user he does not know and on a public blog seemed totally out of place.

I have known many software developers with their same own personal interrelation problems.

Takes a shower every morning and changes his underwear often. Relax, go to the park for jogging or feed the squirrels. In the evening, leave the house, go to the cinema or have a pizza with friends. If you have no friends it seems to me there are sites on internet  that can help you.

Follow my simple guidelines and you will see that over time your problems can be solved.

Regards

lordofhyphens, 2021-08-14:
@atomaticSrl, writing an issue on the tracker (in the first post) equivalent to asking someone to implement it. There is no shortage of people who contribute partially-formed ideas with no expectation of performing any work to realize those ideas.

As I strongly disagree that your proposal is worth my time to implement or further investigate, I was laying out the route for which, should you be wanting your feature as you laid it out, it may be realized. This was my honesty to you and a marker to others who may see your post in the future and wonder why it was not implemented.

Replying with an assumption on my character and interpersonal skills reads as a condescending insult to me. 

I hope that the existing functionality with modifier meshes suits your needs. If it does meet your needs, feel free to close this issue. Should your decide to pursue your proposal, please bring a comprehensive vision of what it is supposed to do, how that complexity is to be presented to the user, and how it may be verified to do what it is supposed to do.

I will review a pull request with your feature.

bubnikv, 2021-08-20:
@atomaticSrl 

> The curse and the arrogance of his response towards a user he does not know and on a public blog seemed totally out of place.
>
> I have known many software developers with their same own personal [...]

Thanks for sharing the recommendations your therapist gave to you. These recommendations may indeed be useful to others.