slic3r/Slic3r #5097: LImits for the layer Height

ZbynekBezchleba, 2021-08-09:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

ZbynekBezchleba, 2021-08-09:
I would like to use slic3r for the generating of  G-code . This code should be used for the printing of concrete objects by robots. The dimensions of such object are approx. 2m x 2 m x 2 m . I would like to set layer height in the range 1- 100 mm/layer, I am not able to reach this goal . When I see the generated G- code I see the Z- axis step is max. 1 mm.  Can you help me with the setting for this intention ?

supermerill, 2021-08-09:
Doesn't seems to have an hardcoded limit.
What is your nozzle diameter?

ZbynekBezchleba, 2021-08-09:
I will be approx. 30 mm for the nozzle diameter

supermerill, 2021-08-09:
my bad, it seems that there is a hardcoded 1mm limit. I don't know where.

ZbynekBezchleba, 2021-08-09:
It means, there is no possibility how to set the higher layer's height than
1 mm ?  Any idea how to do it ?