slic3r/Slic3r #5098: Compile failure

VanessaE, 2021-08-21:
This is using the 

**Describe the bug**

I had to reinstall my system today (for reasons unrelated to 3d printing).  When it came time to compile Slic3r, it made it to about 70%, then died.  I tried googling for the error and came up empty.  Nothing I found made any sense to me.

The failure is shown below:

```cmake
[ 70%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/SLA/Clustering.cpp.o
/home/vanessa/RepRap/Slic3r/src/libslic3r/SLA/IndexedMesh.cpp: In member function ‘Slic3r::sla::IndexedMesh::hit_result Slic3r::sla::IndexedMesh::query_ray_hit(const Vec3d&, const Vec3d&) const’:
/home/vanessa/RepRap/Slic3r/src/libslic3r/SLA/IndexedMesh.cpp:135:23: warning: ‘hit.igl::Hit::id’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  135 |         ret.m_face_id = hit.id;
      |         ~~~~~~~~~~~~~~^~~~~~~~
[ 70%] Linking CXX static library liblibslic3r.a
[ 70%] Built target libslic3r
Scanning dependencies of target libslic3r_gui
[ 70%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx.gch
In file included from /usr/include/wx-3.1-unofficial/wx/platform.h:503,
                 from /usr/include/wx-3.1-unofficial/wx/defs.h:45,
                 from /usr/include/wx-3.1-unofficial/wx/event.h:14,
                 from /usr/include/wx-3.1-unofficial/wx/app.h:19,
                 from /home/vanessa/RepRap/Slic3r/src/slic3r/pchheader.hpp:95,
                 from /home/vanessa/RepRap/Slic3r/build/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/wx-3.1-unofficial/wx/chkconf.h:1249:9: error: #error "wxUSE_WEBREQUEST must be defined, please read comment near the top of this file."
 1249 | #       error "wxUSE_WEBREQUEST must be defined, please read comment near the top of this file."
      |         ^~~~~
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/date_time/time_clock.hpp:17,
                 from /usr/include/boost/date_time/posix_time/posix_time_types.hpp:10,
                 from /usr/include/boost/asio/time_traits.hpp:23,
                 from /usr/include/boost/asio/detail/timer_queue_ptime.hpp:22,
                 from /usr/include/boost/asio/detail/deadline_timer_service.hpp:29,
                 from /usr/include/boost/asio/basic_deadline_timer.hpp:25,
                 from /usr/include/boost/asio.hpp:25,
                 from /home/vanessa/RepRap/Slic3r/src/slic3r/pchheader.hpp:64,
                 from /home/vanessa/RepRap/Slic3r/build/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:83: src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx.gch] Error 1
make[1]: *** [CMakeFiles/Makefile2:1360: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

This is using Wx 3.1.4 from Codelite's repository, since Debian still only supplies 3.0.x (they still have no plans to update, because 3.1.x is development/unstable).  I installed 3.1.4 thusly:

```bash
echo "deb https://repos.codelite.org/wx3.1.5/debian/ buster libs" \
>> /etc/apt/sources.list
apt-key adv --fetch-keys https://repos.codelite.org/CodeLite.asc
apt-get update

apt -y install \
  libwxbase3.1-0-unofficial3 libwxbase3.1unofficial3-dev \
  libwxgtk3.1-0-unofficial3 libwxgtk3.1unofficial3-dev wx3.1-headers \
  wx-common libwxgtk-media3.1-0-unofficial3 libwxgtk-media3.1unofficial3-dev \
  libwxgtk-webview3.1-0-unofficial3 libwxgtk-webview3.1unofficial3-dev \
  libwxbase3.1-0-unofficial3-dbg libwxgtk3.1-0-unofficial3-dbg \
  libwxgtk-webview3.1-0-unofficial3-dbg libwxgtk-media3.1-0-unofficial3-dbg \
  wx3.1-i18n wx3.1-examples
```

**Desktop:**
 - OS: Debian 11, "bullseye", installed today from the 2021-08-18 netinstall image.
 - Version: `merill-merge` branch, commit f88f22445 (using that branch since that's where SuperSlicer is, more or less)

supermerill, 2021-08-21:
I have no problem building it with the BuildLinux.sh script on ubuntu 20.04.
```
./BuildLinux.sh -dsi
```

VanessaE, 2021-08-21:
Not here.  When I re-clone and try to use that script, it dies almost immediately, because other scripts it calls aren't where it expects:

```
vanessa@rainbird:~/RepRap/Slic3r$ ./BuildLinux.sh 
FOUND_GTK2=ii  libgtk2.0-dev:amd64          2.24.33-2    amd64        development files for the GTK library)
[1/9] Updating submodules...
~/RepRap/Slic3r/resources/profiles ~/RepRap/Slic3r
Submodule 'resources/profiles' (https://github.com/slic3r/slic3r-profiles.git) registered for path './'
Submodule 'src/ArcWelderLib' (https://github.com/FormerLurker/ArcWelderLib.git) registered for path '../../src/ArcWelderLib'
Cloning into '/home/vanessa/RepRap/Slic3r/resources/profiles'...
Cloning into '/home/vanessa/RepRap/Slic3r/src/ArcWelderLib'...
Submodule path './': checked out '118aa919c16837eb2ff6ba97e2934fa4144ef806'
Submodule path '../../src/ArcWelderLib': checked out 'fe18f18b4c005a2c4e4b69a5866d0d065bde82aa'
~/RepRap/Slic3r
[2/9] Changing date in version...
done
chmod: cannot access '/home/vanessa/RepRap/Slic3r/build/src/BuildLinuxImage.sh': No such file or directory
[9/9] Generating Linux app...
~/RepRap/Slic3r/build ~/RepRap/Slic3r
./BuildLinux.sh: line 150: /home/vanessa/RepRap/Slic3r/build/src/BuildLinuxImage.sh: No such file or directory
done
vanessa@rainbird:~/RepRap/Slic3r$ find ./ |grep -i build |grep -i image
./src/platform/osx/BuildMacOSImage.sh.in
./src/platform/unix/build_appimage.sh.in
./src/platform/unix/BuildLinuxImage.sh.in

```

lordofhyphens, 2021-08-21:
Did you move the source without clearing the build directory or cache?

VanessaE, 2021-08-21:
Ignore that last comment, I missed the `-dsi` switch. When I run the command *properly* :stuck_out_tongue: it.... works!

`./BuildLinux.sh -dsi` built Slic3r without a single complaint, and it seems to run just fine.

But..that raises the question, why does that work, but my usual build script shown below does not?

```bash
#!/bin/bash

# make sure Slic3r repo is on the `merill-merge` branch

#export CFLAGS="-DDEBUG_SERIAL"

cd /home/vanessa/RepRap/Slic3r
make clean
rm -rf build

mkdir build
cd build
cmake .. -DSLIC3R_BUILD_TESTS=OFF -DSLIC3R_GTK=3
make  -j8
```

As an aside, that problem with the window auto-raising itself all the time seems to have sorted itself out.  I assume that's because your build script employs its own Wx-related options, which mine does not.

supermerill, 2021-08-28:
> But..that raises the question, why does that work, but my usual build script shown below does not?

Imo: Your build script doesn't build the deps. So it has to find them in your computer, and if they are not found or they have an incompatible version, the build will fail.

closed 2021-08-28 by VanessaE