slic3r/Slic3r #5103: Not able to load slic3r on Macos, latest m1.  MacOs not able to confirm not malicious

clexp, 2021-09-05:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
"“Slic3r.app” can’t be opened because Apple cannot check it for malicious software."

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Downloads, find .dmg,'
2. Run the .dmg.  
3. Error ""Slic3r.app” can’t be opened because Apple cannot check it for malicious software."
4. Error continues: "This software needs to be updated. Contact the developer for more information. This item is on the disk image “Slic3r-master-latest.dmg”. Safari downloaded this disk image today at 17:19 from dl.slic3r.org."
5. Repeat for all .dmgs in the repo

**3D Model and Slic3r Configuration Export**
Not applicable

**Expected behavior**
To open normally

**Screenshots**
Nothing more to show besides error above

**Desktop (please complete the following information):**
M1 MacBook Pro, 2021, BigSur 11.5.1

**Additional context**
Sorry if I am missing something dumb.

billdwhite, 2021-11-05:
If you go into the security&privacy control panel, you can tell the OS to bypass that warning and open the file

clexp, 2021-11-05:
Ah that's great, That ties in with some other issues I had, all working now, thanks so much.
regards

lordofhyphens, 2021-11-07:
Note that I have no M1 macs and zero interest in giving Apple my money for
the privilege, so any bugs that are Mac-specific are automatically at the
bottom of the pile. I'll look at patches but cannot debug.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

billdwhite, 2021-11-07:
@lordofhyphens: I don't think there is much to debug.  The program will still run, but you have to just get past the restriction warning;  the true fix would be to get the application properly certified/trusted to run a mac which would bypass that warning but it is still usable at the moment (although it does not run on Mac OS Montery; I get a "perl-local quit unexpectedly" since that update and the program dies immediately and is unusable, but that's belongs in a new issue)

lordofhyphens, 2021-11-08:
I've repeatedly tried to get the signing to work correctly but have not had
the time or the modern OSX install to test the entire flow.
Paid them their tax as well.

Apple hates developers with a passion.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.