slic3r/Slic3r #5117: NULL pointer dereference in AMF XML parser (metadata tag without type attribute)

eldstal, 2021-12-26:
### Summary
A crafted AMF XML document can cause a crash due to a NULL pointer dereference during parsing.

### Vulnerable versions
* Slic3r (commit b1a5500f427700ac3dffc0e7d9535ea65f993537)

### Step to reproduce
1. Create the proof-of-concept OBJ file (`segv_strlen_prolog.amf.xml`):
```
<amf unit="millimeter">
	<object id="0">
		<metadata>dummy</metadata>
	</object>
</amf>
```
2. Execute `slic3r --info segv_strlen_prolog.amf.xml`
3. Observe segmentation fault.

### Cause
An attempt is made to read the `type` attribute of the `<metadata>` tag, at [AMF.cpp:189](https://github.com/slic3r/Slic3r/blob/b1a5500f427700ac3dffc0e7d9535ea65f993537/xs/src/libslic3r/IO/AMF.cpp#L189). The PoC contains a `metadata` tag without a `type` attribute.

`get_attribute()` returns NULL, and thus the creation of the `std::string m_value[0]` crashes.

### Impact
Denial of Service.

### Proposed mitigation
Check for NULL before trying to construct the `std::string`, set a default value or reject the tag. Similar checks are already in place at [line 163](https://github.com/slic3r/Slic3r/blob/b1a5500f427700ac3dffc0e7d9535ea65f993537/xs/src/libslic3r/IO/AMF.cpp#L163) and others.

eldstal, 2022-01-25:
This vulnerability has been assigned [CVE-2021-45846](https://www.cve.org/CVERecord?id=CVE-2021-45846).

supermerill, 2022-02-01:
note that patch should also be applied to prusaslicer @bubnikv