slic3r/Slic3r #5119: NULL pointer dereference in 3MF XML parser (triangle tag without v1/v2/v3 attribute)

eldstal, 2021-12-26:
### Summary
A crafted 3MF XML document can cause a crash due to a NULL pointer dereference during parsing.

### Vulnerable versions
* Slic3r (commit b1a5500f427700ac3dffc0e7d9535ea65f993537)

### Step to reproduce
1. Create the proof-of-concept OBJ file (`3dmodel.3dmodel`):
```xml
<model>
  <resources>
    <object id="1">
      <mesh>
        <vertices>
          <triangle />
        </vertices>
      </mesh>
    </object>
  </resources>
</model>
```
2. Pack the file into a zip archive together with the prerequisite other files from a 3mf file:
```
3D/3dmodel.3dmodel
rels/.rels
[Content_Types].xml
```
3. Rename the zip archive to `nullptr_3mf_triangle.3mf`
4. Execute `slic3r --info nullptr_3mf_triangle.3mf`
5. Observe segmentation fault.

### Example file

[nullptr_3mf_triangle.zip](https://github.com/slic3r/Slic3r/files/7777538/nullptr_3mf_triangle.zip)

### Cause
`get_attribute()` in TMF.cpp returns NULL if the sought attribute is missing. The NULL check at [TMF.cpp:590](https://github.com/slic3r/Slic3r/blob/b1a5500f427700ac3dffc0e7d9535ea65f993537/xs/src/libslic3r/IO/TMF.cpp#L590) is ineffective, since `self->stop()` does not terminate the current function.

Execution continues to line 593, where `atoi` receives a NULL pointer input, and a crash results.

### Impact
Denial of Service.

### Proposed mitigation
Throw an exception in `TMFParserContext::stop()` to ensure that file parsing stops immediately.

eldstal, 2022-01-25:
This vulnerability has been assigned [CVE-2021-45847](https://www.cve.org/CVERecord?id=CVE-2021-45847).

supermerill, 2022-01-31:
not present in merill-merge branch