slic3r/Slic3r #5125: Splash Screen

golavr, 2022-01-30:
**Feature Request**
It's highly recommended to add splash screen to give the user indication that the app is opened and doing some calculations/loading.

**The solution I'd like**
Add splash screen when app starts while having some animation that we are progressing.
It would be nice if the logo would glow or spin or sliced back and forth.

**Alternatives I've considered**
A simple phase 1 solution will be changing text with loaded modules or just loading... blink.

[image]

golavr, 2022-02-18:
I'm closing my issue, this project is dead. I found that [superslicer](https://github.com/supermerill/SuperSlicer) having splash and really nice one, it also has many features and cool UI.

closed 2022-02-18 by golavr

lordofhyphens, 2022-02-19:
Everyone wants something done; nobody wants to do it.