slic3r/Slic3r #5134: Only 1/4 of the bed showing

psavonen, 2022-04-02:
[image]
Picture tells it all. The problem doesn't appear on 2D view.

Running 4K monitor on Ubuntu 21.10. Slic3r version 3.

**3D Model and Slic3r Configuration Export**
Slic3r_config_bundle.zip](https://github.com/slic3r/Slic3r/files/8402267/Slic3r_config_bundle.zip)

Desktop :
 - OS: Ubuntu
 - Version 21.10
  -Version 3, can't really see the actual release version
[image]

ComradeCon, 2023-06-04:
Did you figure this out? I am having the same problem. Running Linux Mint 21.1 Cinnamon.

ComradeCon, 2023-06-04:
I figured out the problem. If you have monitor scaling set to anything but 100% the window only takes up 1/4 of the screen.