slic3r/Slic3r #5137: Reading color attributes for vertices from obj files

balloonhotair, 2022-04-07:
Has anyone tried to accessed the color info for the vertices within obj files to create gcode filament changes? Meshlab can take ply files with color attributes for the vertices and save the files as obj with the color info intact so I was looking to see if I can modify the slicer to do the filament changes when producing the gcode.  If there is anybody who could point me to the files that read in obj files would be big time saver.  I see there is a gcode writer class so that should be the other end of my idea. I did a fair amount of C++ years back so the coding part I know, it is just getting a feel for where all the pieces are in github that I would like to make quicker and anything to be aware of if I do this.

lordofhyphens, 2022-04-07:
Don't believe we have. That would be under libslic3r's src.

https://github.com/slic3r/Slic3r/tree/merill-merge/src/libslic3r

You want the Format classes, things that touch TriangleMesh.

Color in the UI isn't used in slicing (it's a Material property).

You probably should subdivide the mesh into its components based on color
if you wanted to get this through with the least amount of pain and
suffering.

lordofhyphens, 2022-04-07:
You should target that branch for your changes.

balloonhotair, 2022-04-07:
Thanks you for giving me a starting point.

supermerill, 2022-05-06:
check the GLGizmoMmuSegmentation::update_model_object()
it shows how to set the ModelObject's ModelVolume's mmu_segmentation_facets

I don't 100% understand the code. The data object is a bit cryptic (usage is in TriangleSelector::deserialize, but I don't understand it fully)