slic3r/Slic3r #5139: how to export a new .stl file from a old one by command line

link577, 2022-04-26:
### Version
1.3.0

I want change some config and generate a new stl file(example_new.stl) from the old one(example_old.stl) by command line, such as :

$ slic3r --fill-density 70 example_old.stl

I change the fill density to 70%. But I don't know how to export a new stl file. In version 1.3.0 "--export-stl" command is not useful. Can anyone helps me. Thanks so much!

lukewarm-h2o, 2022-08-31:
--fill-density is a slicer setting and wont affect the triangle mesh data in an stl.
If you are trying to save print configurations:
"
You can also create a config file from[ command line:](https://manual.slic3r.org/advanced/command-line)

slic3r --nozzle-diameter 0.35 --filament-diameter 2.85 \
    --temperature 185 --first-layer-temperature 195 --layer-height 0.2 \
    --save my_config.ini
"

link577, 2022-09-01:
> --fill-density is a slicer setting and wont affect the triangle mesh data in an stl. If you are trying to save print configurations: " You can also create a config file from[ command [...]

I see，thanks！

closed 2022-09-01 by link577