slic3r/Slic3r #5142: dl.slic3r.org is down. Cant access automated builds

vascojdb, 2022-07-07:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
Cant access https://dl.slic3r.org/ to download automated builds.
I am formatting my PC and wanted to download the same version I had. (or at least a newer development one.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://dl.slic3r.org/'
2. See error: This site can’t be reached

**Expected behavior**
Website works

**Desktop (please complete the following information):**
 - OS: Windows

mwelchuk, 2022-07-27:
Probably wort noting that the inaccessibility extends to:

- https://dl.slic3r.org/linux/
- https://dl.slic3r.org/mac/
- https://dl.slic3r.org/win/

i.e. All the main download links provided on the download page on the website...

lordofhyphens, 2022-07-28:
@alranel it's your server

3dprogramin, 2022-08-11:
Still down, the entire subdomain https://dl.slic3r.org

rebz, 2022-08-21:
Still down. Unable to download any version of the program.

foreachthing, 2022-09-15:
Still down... 😦

CarrollDiehl, 2022-12-21:
Can someone provide me a current pre-built Linux version? I'm running into some errors when trying to build it myself. Thanks! (: